JoelBender/bacpypes #360: Getting Raspberry pi to act as a Bacnet device

martinwaroe, 2020-10-07:
Hello,
I dont know is this is the correct way to ask my question so feel free to direct me elsewhere.
I have looked trough the documentation and have som questions.

My goal is to have a raspberry pi run Bacnet on a closed network. Let's say i have a network with 5 ip addresses. 
10.0.20.2 to 10.0.20.6. with mask of 255.255.255.0 and a gateway with 10.0.20.1.
I want the raspberry to broadcast analog values. This values is going to be generated from another python script i have so no physical I/O's.

I then want to be able to read the raspberry pi as a device in YABE or other bacnet scanners from another computer via ethernet port on the pi. My computer will be connected via the same IP range. Is this possible with this module?

Another question i have:
When trying the sample scripts provided i get a bit confused with how the .ini file should look like.
as mentioned above i am connected to the pi with a laptop via ethernet. we are in the same ip range (IP: 10.0.20.3).

the .ini file i have created looks like this:

**[BACpypes]
objectName: Raspberry
address: 10.0.20.2 _Is this the pysical ipaddress on the pi?_
objectIdentifier: 599
maxApduLengthAccepted: 1024
vendorIdentifier: 15
foreignBBMD: 10.0.20.3 _Should this be the ip of the laptop connected to the pi?_
foreignTTL: 30**

Hope you can take time to answer some of my questions. I'm looking forward to working with the Bacpypes module.

JoelBender, 2020-10-07:
In this scenario, your pi is running a BACpypes application that will be sending out broadcast change-of-value notifications which you should be able to receive in YABE.  As a client, YABE or some other BACnet application will be able to send it Read Property requests to see what the "present value" is of the analog values.

Your INI file must have **address: 10.0.20.2/24** which is the IP address and subnet mask together in [CIDR notation](https://en.wikipedia.org/wiki/Classless_Inter-Domain_Routing), the **objectIdentifer** must be unique within the entire BACnet network.  You will not be using the "foreign device" portion of BACnet/IPv4, so the last two settings can be omitted.

Try starting out with running the `Sample4_RandomAnalogValueObject.py` on the pi and see what you can read from your other BACnet clients, then try running the `ReadProperty.py` sample application and see values from there.

martinwaroe, 2020-10-08:
Thanks you for the fast response.
Running the `Sample4_RandomAnalogValueObject.py` gives me some errors.
IP on the raspberry pi is: 10.0.20.2. mask: 255.255.255.0. Gateway: 10.0.20.1
`BACpypes.ini` file is located in the same folder as the sample. and looks like this:

[BACpypes]
objectName: Raspberry
address: 10.0.20.2/24
objectIdentifier: 399
maxApduLengthAccepted: 1024
segmentationSupported: segmentedBoth
vendorIdentifier: 15

Error I get when running from my IDE (Thonny)

`>>> %Run Sample4_RandomAnalogValueObject.py
Traceback (most recent call last):
  File "/home/pi/Downloads/bacpypes-master/samples/HandsOnLab/Sample4_RandomAnalogValueObject.py", line 121, in <module>
    main()
  File "/home/pi/Downloads/bacpypes-master/samples/HandsOnLab/Sample4_RandomAnalogValueObject.py", line 100, in main
    this_application = BIPSimpleApplication(this_device, args.ini.address)
  File "/usr/local/lib/python3.7/dist-packages/bacpypes/app.py", line 535, in __init__
    self.mux = UDPMultiplexer(self.localAddress)
  File "/usr/local/lib/python3.7/dist-packages/bacpypes/bvllservice.py", line 96, in __init__
    self.directPort = UDPDirector(self.addrTuple)
  File "/usr/local/lib/python3.7/dist-packages/bacpypes/udp.py", line 155, in __init__
    self.bind(address)
  File "/usr/lib/python3.7/asyncore.py", line 328, in bind
    return self.socket.bind(addr)
OSError: [Errno 99] Cannot assign requested address`

JoelBender, 2020-10-08:
There are a few possible problems; 1) the IP address is incorrect, 2) there is another application running that already has UDP port 47808 open, or 3) there is a firewall rule blocking the application from opening the port (similar to the notion that you have to have special privileges to open port numbers less than 1024).  My guess is that you have been careful with (1) but didn't realize that there is already a service/process using the port (2).

martinwaroe, 2020-10-11:
1) The IP address should be correct. The .INI file and Raspberry pi have the same address (10.0.20.2/24).
2)How would you go about finding out if another service/process is running the 47808 port?
3) I am running Raspbian. This does not come with a firewall at all. so that should not be the problem.

I tried defining another port(47809) in .ini file but same error output.

JoelBender, 2020-10-11:
On older Ubuntu distributions you can use `netstat --udp -a -n | grep 47808`, newer distributions have replaced it with `ss`:
```
$ ss --udp -a -n -p | grep 47808
UNCONN   0     0            10.0.1.255:47808        0.0.0.0:*    users:(("python3.6",pid=4742,fd=4))
UNCONN   0     0             10.0.1.90:47808        0.0.0.0:*    users:(("python3.6",pid=4742,fd=3))
```
Notice that the same process has both the unicast address and broadcast address bound.

You need to have two different INI files for the two applications, they need separate socket numbers and device instance numbers, so one might have the address `1.2.3.4/24:47808` and the other will be `1.2.3.4/24:47809`.

martinwaroe, 2020-10-13:
Nothing was using the port. So I tried changing my WIFI IP to match the config. That worked. I was wondering if there is a file/place where I need/can define what connection (WIFI/Ethernet) the INI file should point at.

JoelBender, 2020-10-13:
The `ip addr` command gives you a list of interfaces and associated IP addresses so in order to [bind](https://docs.python.org/3.8/library/socket.html#socket.socket.bind) the communications socket the address has to be one of them.  You can have multiple INI files with different addresses in each one and pass the file name as an option to the application like:
```
$ python some_application.py --ini addr1.ini
```
Note that you can have multiple IPv4 networks on the same physical interface (a.k.a. single link multi-homing) or on multiple interfaces (like your WIFI and Ethernet).  If you are using DHCP so you don't get the same IP address every time (so called "static DHCP") then you might want to install the [netifaces](https://pypi.org/project/netifaces/) library, BACpypes will use it to resolve the interface IP address given the name so your INI file can contain **address: eth1**.

martinwaroe, 2020-10-13:
Do I need to `import netifaces` in the `Sample4_RandomAnalogValueObject.py` or does it gather the eth0 config automatically?

martinwaroe, 2020-10-13:
netifaces worked like a sharm.

closed 2020-10-13 by martinwaroe