JoelBender/bacpypes #361: Updating presentValue externally in running COVServer

JimitD, 2020-10-08:
Hi Joel, 

I am using COVServer to create a BACnet device. 

I would like to update the presentValue of Object externally without using recurring task. So, can I access Object created in COVServer externally to change it's presentValue ? 

Thanks in advance.

JoelBender, 2020-10-08:
What mechanism or protocol would you expect to use to tell the server process to change the value?  Options would be things like a web browser page with some UI controls like buttons and sliders, a web service interface where you can GET the current value and POST a JSON blob requesting the value to change, subscribe to an MQTT topic where other broker clients can post new values, etc.

JimitD, 2020-10-09:
I would like to use another python script on same machine where COVServer is running to access objects created by COVServer. 
So, can I make any request to COVServer to change value? Any small example would be appropriated.

JoelBender, 2020-10-10:
Rather than layer on top some kind of remote procedure call protocol, the simplest answer is to just make the `presentValue` writable and use BACnet.  In the commit I just made to the `COVServer.py` sample it makes that change.  Now in another process run the `ReadWriteProperty.py` sample application enter something like `write <addr> analogValue:1 presentValue 12.3`.

If the two applications are on the same device they will have the same IP address so they will need separate port numbers, and it's usually easier to keep the server at 47808 and other types of clients at 47809, 47810, etc.

JimitD, 2020-10-10:
Hi Joel, 

Thank you for inputs. But, this makes PV changeable with third party client too. I don't want to give access to any third party client app to change PV.  

Your solution with two different port address suffices the need but makes enable third party to change it too. 

How can I restrict in such case?

JoelBender, 2020-10-11:
I still think that you should stick to BACnet rather than layering any one of the hundreds of process-to-process communications techniques that are available, so just checking the source address of the `WritePropertyRequest` might be good enough:
```
    def do_WritePropertyRequest(self, apdu):
        """Change the value of some property of one of our objects."""
        if _debug: SubscribeCOVApplication._debug("do_WritePropertyRequest %r", apdu)

        # check the source address
        if apdu.pduSource != ("1.2.3.4", 47809):
            raise ExecutionError(errorClass='property', errorCode='writeAccessDenied')

        return super().do_WritePropertyRequest(apdu)
```

JimitD, 2020-10-12:
That's sound perfect. 

Thank you Joel for wonderful work.

closed 2020-10-12 by JimitD