JoelBender/bacpypes #363: Updating Analog values every 10 minutes

martinwaroe, 2020-10-24:
Hello

I am using the `RandomAnalogValueObject.py` to generate Analog values to YABE from a .Json file.
I have two issues/questions.

1:
Is there a way to generate individual analog values within the sample? I would like to be able to add more values. And I want the value to be different for each point.

2: 
Do the Bacpypes module have something to refresh the values output to for instance  YABE. It would be helpful to update the values every 10 minutes.

Heres the code i have so far:

AnalogValue.py:

```
import os
import time
import json

from bacpypes.debugging import bacpypes_debugging, ModuleLogger
from bacpypes.consolelogging import ConfigArgumentParser

from bacpypes.core import run

from bacpypes.primitivedata import Real
from bacpypes.object import AnalogValueObject, Property, register_object_type
from bacpypes.errors import ExecutionError

from bacpypes.app import BIPSimpleApplication
from bacpypes.local.device import LocalDeviceObject

# Read Json
with open('/Users/martin/python/projectX/ObjectList.json') as json_file:
    data = json.load(json_file)

    def get_present_value(no):
        for a in data['AnalogValues']:
            if a['ObjectIdentifier'] == int(no):
                return a['PresentValue']
        return None

ai_1 = (get_present_value(1))

print(ai_1)

# some debugging
_debug = 0
_log = ModuleLogger(globals())

# settings
RANDOM_OBJECT_COUNT = int(os.getenv('RANDOM_OBJECT_COUNT', 1))

#
#   RandomValueProperty
#

class RandomValueProperty(Property):

    def __init__(self, identifier):
        if _debug: RandomValueProperty._debug("__init__ %r", identifier)
        Property.__init__(self, identifier, Real, default=0.0, optional=True, mutable=False)

    def ReadProperty(self, obj, arrayIndex=None):
        if _debug: RandomValueProperty._debug("ReadProperty %r arrayIndex=%r", obj, arrayIndex)

        # access an array
        if arrayIndex is not None:
            raise ExecutionError(errorClass='property', errorCode='propertyIsNotAnArray')

        # return a random value

        value = ai_1

        if _debug: RandomValueProperty._debug("    - value: %r", value)

        return value

    def WriteProperty(self, obj, value, arrayIndex=None, priority=None, direct=False):
        if _debug: RandomValueProperty._debug("WriteProperty %r %r arrayIndex=%r priority=%r direct=%r", obj, value,
                                              arrayIndex, priority, direct)
        raise ExecutionError(errorClass='property', errorCode='writeAccessDenied')

bacpypes_debugging(RandomValueProperty)

#
#   Random Value Object Type
#

class RandomAnalogValueObject(AnalogValueObject):
    properties = [
        RandomValueProperty('presentValue'),
    ]

    def __init__(self, **kwargs):
        if _debug: RandomAnalogValueObject._debug("__init__ %r", kwargs)
        AnalogValueObject.__init__(self, **kwargs)

bacpypes_debugging(RandomAnalogValueObject)
register_object_type(RandomAnalogValueObject)

#
#   __main__
#

def main():
    # parse the command line arguments
    args = ConfigArgumentParser(description=__doc__).parse_args()

    if _debug: _log.debug("initialization")
    if _debug: _log.debug("    - args: %r", args)

    # make a device object
    this_device = LocalDeviceObject(
        objectName=args.ini.objectname,
        objectIdentifier=('device', int(args.ini.objectidentifier)),
        maxApduLengthAccepted=int(args.ini.maxapdulengthaccepted),
        segmentationSupported=args.ini.segmentationsupported,
        vendorIdentifier=int(args.ini.vendoridentifier),

    )

    # make a sample application
    this_application = BIPSimpleApplication(this_device, args.ini.address)

    # make some random input objects
    for i in range(1, RANDOM_OBJECT_COUNT + 1):
        ravo = RandomAnalogValueObject(
            objectIdentifier=('analogValue', i),
            objectName='Temp%d' % (i,),

        )

        this_application.add_object(ravo)

    # make sure they are all there

    _log.debug("    - object list: %r", this_device.objectList)

    _log.debug("running")

    run()

    _log.debug("fini")

if __name__ == "__main__":
    main()
```
Heres my .Json file:

```
{
    "AnalogValues": [
        {
            "ObjectIdentifier": 1,
            "PresentValue": 59
        },
        {
            "ObjectIdentifier": 2,
            "PresentValue": 60
        }
    ]
}
```

SQ9MDD, 2020-10-29:
You can look at this  great sample:
[](https://github.com/JoelBender/bacpypes/blob/792041b3ca103c85d7dfabd81f1f61bae5efdba3/samples/mini_device.py)

You can build device with Analog Values and update values in intervals.

martinwaroe, 2020-10-29:
Very useful sample. Thank you!

closed 2020-10-29 by martinwaroe