JoelBender/bacpypes #369: AnalogValueObject.presentValue is not writeable

goffioul, 2020-11-13:
Using mini_device.py as template, I created a simple BACnet device simulator that contain an analogValue object. However, the presentValue is not writeable (https://github.com/JoelBender/bacpypes/blob/master/py34/bacpypes/object.py#L1037). Should it be made writeable instead? By comparison, BinaryValueObject.presentValue is writeable.

JoelBender, 2020-11-13:
I'm not sure why the BinaryValueObject is writable, the definition in the standard is R(1), where "This property is required to be writable when Out_Of_Service is TRUE," it probably is left over from experiments with that.  These object definitions are more like a schema that provides the way to find out the expected property datatype.  In your code you can provide your own AnalogValueObject and override the property definition:
```
from bacpypes.primitivedata import Real
from bacpypes.object import register_object_type, WritableProperty, AnalogValueObject as _AVO

@register_object_type
class AnalogValueObject(_AVO):
    properties = [ WritableProperty('presentValue', Real) ]

avo = AnalogValueObject(presentValue=75.3)
avo.WriteProperty('presentValue', 99.0)
```
I wonder how many other applications assume that BinaryValueObject.presentValue is writable when it shouldn't be :-(.

jahm86, 2020-12-10:
> I'm not sure why the BinaryValueObject is writable, the definition in the standard is R(1), where "This property is required to be writable when Out_Of_Service is TRUE," it probably is left over [...]

Thank you for providing this explanation and workaround. Lately, i saw this after writing an issue.

But i also saw in many control equipments that objects that ends with 'ValueObject' are writable. So i don't understand why they don follow the standards, as you explained.

ChristianTremblay, 2020-12-10:
Not sure at what extent there is confusion between presentValue being writable and BV providing a priorityArray....