JoelBender/bacpypes #370: Cant get parameter right from ini file

sansal54, 2020-11-23:
I try to use misty wrapper library to communicate with mstp network. When I define configurtion file as below :

```
[BACpypes]
objectName: BACClient
; MSTP Local address
address: 25
; The serial port device
interface: /dev/ttySC0
; other mstp config parameters max_masters, baudrate, maxinfo
max_masters: 127
baudrate: 76800
maxinfo: 1
objectIdentifier: 599
maxApduLengthAccepted: 1024
segmentationSupported: segmentedBoth
vendorIdentifier: 15
foreignPort: 0
foreignBBMD: 128.253.109.254
foreignTTL: 30
```
I get such error :

```
Initialized the socket
mac_address = 25 
max master = 127
baud rate = 76800 
max info frames = 1 
/: Is a directory
RS485: Initializing /
```
My RS485 port is ok. I checked it with other devices and communication protocols. I tried the find the problem in bacpypes.local.device - LocalDeviceObject but there was no succeed.
```
    # make a device object
    mstp_args = {
        '_address': int(args.ini.address),
        '_interface':str(args.ini.interface),
        '_max_masters': int(args.ini.max_masters),
        '_baudrate': int(args.ini.baudrate),
        '_maxinfo': int(args.ini.maxinfo),
    }

    this_device = LocalDeviceObject(ini=args.ini, **mstp_args)
```
So I need a help here. Why I get "/: Is a directory" while it should try to connect to "/dev/ttySC0"? It sortof sees only "/" part.

JoelBender, 2020-11-24:
I'm working my way through this, but this is new territory, and the `/: Is a directory` isn't coming from BACpypes.  Inside the MSTPDirector it is looking for an `_mstp_dir` attribute of the device object, but I don't understand how that is related to the `_interface` attribute.  If you can build the library and provide the `-DDEBUG` option it should turn on lots of debugging output from the MSTP agent (in `mstp_agent.c`).

Please open a new issue with misty and reference this issue, I would like to see how it gets resolved.

WeiTangLau, 2021-11-02:
Any update on this?

JoelBender, 2021-11-03:
You need to open a [new issue with misty](https://github.com/riptideio/misty/issues) if you haven't yet.