JoelBender/bacpypes #372: IP2VLANRouter.py

lostianae, 2020-11-28:
Router itself is not discovered as device on its network, only VLAN slaves are discovered.... is it possible to see also router as device?

JoelBender, 2020-11-28:
The way this particular sample has been created, no, there is no device object for the router, it only has the network layer and does not have an application layer.  This is a very old school way of looking at routers, the more modern way is to have an application layer (which would include the device object) and network port objects for each port.  According to the BTL Implementation Guidelines...

> ... there is currently a proposal before the BACnet committee to require routers to have application layers.

lostianae, 2020-11-28:
Thank you for response.

…one more question, what do you think is a maximum of devices emulated by one ip like this router example?

Network slave limit seems to be 256, but when you spawn more network ids, it works for a while, maybe up to 600 slaves and then bacpypes or scanner baceye on the other side simply stops working.

JoelBender, 2020-11-29:
The devices on the VLAN use single octet addresses, so you should have gotten an error attempting to call `Address(257)` for example.  However, if you build the addresses with a larger address space like `vlan_address = Address(struct.pack(">I", device_number))` then you can build lots of them.  Note that every device sends out a global broadcast **I-Am** which will not only appear on the IP side, but will also be processed by every VLAN device, sending and receiving global broadcast messages is going to suffer a performance hit.

If you run the application with many VLAN devices, check what the application is doing in Wireshark or with debugging turned on and let it go through all the startup Who-Is/I-Am processing.  With 1000 devices it could take five to ten minutes.