JoelBender/bacpypes #375: Accessing BACnet device over VPN

JimitD, 2020-12-15:
Hi Joel,

I was have created a BACnet device with local IP address. Which is in range 192.168.X.XX 

This device has OpenVPN running on it with IP address of 172.30.XX.XX. 

So, by VPN connection, remotely I would like to Read Properties of BACnet device. 

Please suggest.

JoelBender, 2020-12-15:
This will be simpler if you have the ability to run BACpypes applications on each side of the tunnel.  The basic idea is to send very simple UDP packets between the two devices based on their IP addresses and see how the source address is presented when it is received.  Assuming that you have your **A** device locally and your **B** device is on the destination network with other BACnet devices.

Install BACpypes on both machines and run the `samples/UDPConsole.py` application.  In both cases you give the application the IP address and port number, so on the **A** device you would give it `172.30.XX.XX/16` because this is a 16-bit network in CIDR notation, and use the default port 47808.  On the **B** device give it `172.30.YY.YY/16`, assuming that makes sense, and it might not!

With both applications running, enter `172.30.YY.YY spam` in the **A** device and see what the source address is when it arrives on the destination.  This is a unicast message.

Try again by entering `* eggs` on the **A** device and see if the VPN tunnel will (1) recognize that the packet is being sent to a broadcast address, and (2) will forward the packet to the VPN server at the other end that will (3) re-broadcast the packet with the same source address.  If this works you are golden, the rest of the BACnet work is easy.  If this doesn't work, there's more work that needs to be done.

Let me know what you find out.

JoelBender, 2020-12-16:
Just to see how this goes, I'm going to convert this issue to a discussion.

closed 2020-12-16 by JoelBender