JoelBender/bacpypes #379: IP2VLANRouter.py

memanasraza, 2020-12-19:
Hi @JoelBender !
I am running IP2VLANRouter for simulating multiple devices. I tried YABE for detecting them. 
YABE is detecting it but for ReadProperty and ReadPropertyMultiple is giving an error.
How can I solve this?

Thanks.
Muhammad Anas RAza

JoelBender, 2020-12-19:
What is the error?  Turn on debugging for the application layer `--debug bacpypes.app` and look at what is being sent and received.

memanasraza, 2020-12-21:
On sending ReadPropertyRequest, the response is giving the following error:
```
<bacpypes.apdu.ReadPropertyRequest(12,32) instance at 0x28cc2393730>
        pduSource = <RemoteStation 5:0xc0a80a02c7ad>
        pduExpectingReply = True
        pduNetworkPriority = 0
        apduType = 0
        apduSeg = False
        apduMor = False
        apduSA = True
        apduMaxSegs = 7
        apduMaxResp = 5
        apduService = 12
        apduInvokeID = 32
        objectIdentifier = ('device', 602)
        propertyIdentifier = 'deviceAddressBinding'
        pduData = x''
DEBUG:__main__.VLANApplication:[2]response <bacpypes.apdu.Error(12,32) instance at 0x28cc2393520>
    <bacpypes.apdu.Error(12,32) instance at 0x28cc2393520>
        pduDestination = <RemoteStation 5:0xc0a80a02c7ad>
        pduExpectingReply = 0
        pduNetworkPriority = 0
        apduType = 5
        apduService = 12
        apduInvokeID = 32
        errorClass = 'property'
        errorCode = 'unknownProperty'
        pduData = x''
DEBUG:__main__.VLANApplication:[2]indication <bacpypes.apdu.ReadPropertyRequest(12,33) instance at 0x28cc238f820>
    <bacpypes.apdu.ReadPropertyRequest(12,33) instance at 0x28cc238f820>
        pduSource = <RemoteStation 5:0xc0a80a02c7ad>
        pduExpectingReply = True
        pduNetworkPriority = 0
        apduType = 0
        apduSeg = False
        apduMor = False
        apduSA = True
        apduMaxSegs = 7
        apduMaxResp = 5
        apduService = 12
        apduInvokeID = 33
        objectIdentifier = ('device', 602)
        propertyIdentifier = 'databaseRevision'
        pduData = x''
DEBUG:__main__.VLANApplication:[2]response <bacpypes.apdu.Error(12,33) instance at 0x28cc2380070>
    <bacpypes.apdu.Error(12,33) instance at 0x28cc2380070>
        pduDestination = <RemoteStation 5:0xc0a80a02c7ad>
        pduExpectingReply = 0
        pduNetworkPriority = 0
        apduType = 5
        apduService = 12
        apduInvokeID = 33
        errorClass = 'property'
        errorCode = 'unknownProperty'
        pduData = x''
```

memanasraza, 2020-12-23:
Hello @JoelBender 
Hope you are doing well.
I turned on debugging for the application layer.
Get this error:
`DEBUG:bacpypes.app.Application:indication <bacpypes.apdu.ReadPropertyRequest(12,34) instance at 0x1a6ed42a6d0>
    <bacpypes.apdu.ReadPropertyRequest(12,34) instance at 0x1a6ed42a6d0>
        pduSource = <RemoteStation 10:0xc0a80a02e2f4>
        pduExpectingReply = True
        pduNetworkPriority = 0
        apduType = 0
        apduSeg = False
        apduMor = False
        apduSA = True
        apduMaxSegs = 7
        apduMaxResp = 5
        apduService = 12
        apduInvokeID = 34
        objectIdentifier = ('analogValue', 1)
        propertyIdentifier = 'units'
        pduData = x''
DEBUG:bacpypes.app.Application:    - helperFn: <bound method ReadWritePropertyServices.do_ReadPropertyRequest of <__main__.VLANApplication object at 0x000001A6ED3DC610>>
DEBUG:bacpypes.app.Application:    - execution error: ExecutionError('property', 'unknownProperty')`

memanasraza, 2020-12-23:
prople

memanasraza, 2020-12-23:
The issue was I haven't allowed the ReadPropertyMultiple in IPVLANRouter file.
`$ python IP2VLANRouter.py 192.168.x.x 10 20 --rpm --count 5`

closed 2020-12-23 by memanasraza