JoelBender/bacpypes #381: ReadProperty.py on the same device using IP2VLANRouter.py

memanasraza, 2020-12-24:
Hello @JoelBender, @ChristianTremblay. Hope you are doing well.
I have two devices A & B. **DEVICE A** is running IP2VLANRouter.py with 5 and I have run ReadProperty.py and ReadWriteProperty.py samples on **DEVICE B**. 
I can detect the devices from ReadProperty.py pr ReadWriteProperty.py on **DEVICE B** using:
```
>> python samples/ReadProperty.py --ini BAC0.ini
> read 20:2 analogValue:2 presentValue
15.0233
```
My question is;
**- How can I detect them on DEVICE A?**

Thanks in Anticipation.

JoelBender, 2020-12-24:
The IP2VLANRouter.py application doesn't have a "console" or other way of interacting with it while it's running, therefore there is no way to initiate a Read/Write Property Request.  You could add one by merging the ReadWriteProperty.py into it, and to keep it relatively simple assume that your "console" is going to be stacked on top of the first of the five virtual devices.

JoelBender, 2020-12-24:
Try the new additional sample and let me know if it works.

memanasraza, 2020-12-24:
Thank you for the response @JoelBender. 
Tried running above code. `python 192.168.0.118 10 20 --rpm --count 5`
In console, I tried running:
```
> whois
> read 20:2 device:2002 objectName
noResponse
> read 10:192.168.0.118 device:2002 objectName
noResponse
> whois 192.168.0.118:47808 

```
I think I had not clearly explained the question:
I have two laptops on the same network. IP2VLANRouter.py is running on LAPTOP A using `python IP2VLANRouter.py 192.168.0.118 10 20 --rpm` and I am reading the values on LAPTOP B 
```
> read 20:2 device:2002 objectName
20.2224
```
I detect the devices on different LAPTOP or can do it with a VM. My Question is HOW ADDRESSING WILL WORK ON THE SAME LAPTOP? I AM A BEGINNER to BACnet and BACpypes. WHAT CONCEPTS I AM MISSING? 

Thank you for your time.

JoelBender, 2020-12-25:
BACnet is a protocol for device-to-device communications, there is no concept of "send this request to myself".  You can run two different applications on the same machine, like both IP2VLANRouter.py and ReadWriteProperty.py, provided they use different UDP ports.  In this case, you have two different devices with two different BACnet addresses on two different BACnet networks.

Before getting too deep into BACnet routing, here is a simple way to get two applications on the same machine to talk to each other:

1) create an INI file called **BAC0.ini** and put `192.168.0.118/24:47808` in the address field and give it device instance number 1234.
2) in a terminal window, run `python ReadWriteProperty.py --ini BAC0.ini`
3) create a second INI file called **BAC1.ini** and put `192.168.0.118/24:47809` in the address field and give it device instance number 5678.
4) in a second terminal window, run `python ReadWriteProperty.py --ini BAC1.ini`

You now have two applications with two different UDP ports on the same machine and they can talk to each other.

5) in the second window, enter `read 192.168.0.118:47808 device:1234 objectName` and it will return the device name (which is a common shortcut of saying "the object name of the device object in the device").
6) in the _first_ window, enter `read 192.168.0.118:47809 device:5678 objectName` and it will return the other device name.

In both of those cases the application in one window is talking to the application in the other window using BACnet.  What is different about BACnet than what you might be expecting is that the two applications are peers and can exchange content between them in both directions at the same time.

memanasraza, 2020-12-25:
Did the same using IP2VLANRouter.py.
`python IP2VLANRouter.py 192.168.0.118/24:47808 10 20 --rpm --count 5 --debug bacpypes.udp`
Running 
`python ReadWriteProperty.py --ini BAC1.ini` with address 192.168.0.118/24:47809 in the BAC1.ini.

```
> read 192.168.0.118:47808 device:2002 objectName
noResponse
> read 192.168.0.118:47808 device:2002 objectName
noResponse
> read 10:192.168.0.118:47808 device:2002 objectName
noResponse
> read 20:192.168.0.118:47808 device:2002 objectName
noResponse
```

JoelBender, 2020-12-27:
You are not running ReadWriteProperty.py in both windows, see step (2) and (4) above.  The IP2VLANRouter.py application does _not_ have a BACnet Application layer at the IP address, only a network layer.  If you want to get a response from device 2002 in the second window, first tell the application the address of the router to network 20, then read the data:
```
> rtn 192.168.0.118 20
> read 20:2 device:2002 objectName
VLAN Node 2002
```
The reason why you have to tell your second application the address of the router to network 20 is because (1) there is no concept of a central authority for things like a BACnet network topology and device registration and (2) the second application cannot "see" the IP2VLANRouter.py application because they are not in the same broadcast domain.  There is a network layer message called **Who-Is-Router-To-Network** that would be used for this "network path discovery" process.

This is described in **Clause 6** of the standard.

memanasraza, 2020-12-28:
It's working now. 
Thank you @JoelBender

closed 2020-12-28 by memanasraza