JoelBender/bacpypes #384: Write property description with spaces

mosterme1003, 2021-01-25:
When I write a text with spaces to the description property I get an error because the spaces indicate more than one argument for the property value. Is there a way to fix this?

```
bacnet = BAC0.connect(ip=ipLocal)
bacnet.write('192.168.178.10 analogInput 1 description test text')
```
```
File "E:\Python\lib\site-packages\BAC0\core\io\Write.py", line 88, in write
    iocb = IOCB(self.build_wp_request(args, vendor_id=vendor_id))
  File "E:\Python\lib\site-packages\BAC0\core\io\Write.py", line 200, in build_wp_request
    obj_type, obj_inst, prop_id, value, priority, indx = self._parse_wp_args(args)
  File "E:\Python\lib\site-packages\BAC0\core\io\Write.py", line 131, in _parse_wp_args
    indx = int(args[4])
ValueError: invalid literal for int() with base 10: 'text'
```

JoelBender, 2021-01-26:
This is an issue for [BAC0](https://gitter.im/ChristianTremblay/BAC0), I don't see an easy way to cross post this to a different project, so please create a new issue.

JoelBender, 2021-01-26:
@ChristianTremblay - note to you - incoming!

closed 2021-01-26 by JoelBender

mosterme1003, 2021-01-26:
Sorry, I think I got confused with your repos. I wanted to create this issue in BAC0 in the first place.

ChristianTremblay, 2021-01-26:
For reference....as the answer I pretty close to bacpypes 
https://github.com/ChristianTremblay/BAC0/issues/246