JoelBender/bacpypes #390: Problem with INI file and WhoIsIAm

KostBMF, 2021-03-02:
Hello

I'm a new in BacNet, network, and your library
I'm on a Imac. 

I have a Schneider energy meter connected to a BacNet router. 
I'm trying to run the WhoIsIAM python script, but I get an error message in my terminal

socket.error: [Errno 49] Can't assign requested address

IMac Ip : 10.1.14.200
BacNet Router IP : 10.1.14.201        Gateaway 10.1.14.1
SubnetMask : 255.255.255.0

In my INI file, I put in address 10.1.14.201/24 
The Device ID, I found it with  Yabe software, which find my device on the port 47808
When I send a WhoIs Message with Yabe, i saw it on WireShark, and the Iam Answer on my device 
But when i try to use your script, i'v got the error.

ChristianTremblay, 2021-03-02:
The address in the INI file must be the local address where the script runs.

KostBMF, 2021-03-02:
Thank for you reply
But when I wrote the local IP where the script run, I had this message 
➜  samples python WhoIsIAm.py
> No handlers could be found for logger "bacpypes.appservice.ApplicationServiceAccessPoint"

JoelBender, 2021-03-03:
The **ifconfig** command will tell you your local IP address and subnet mask:
```
en8: flags=8863<UP,BROADCAST,SMART,RUNNING,SIMPLEX,MULTICAST> mtu 1500
        ...
        inet 10.0.1.23 netmask 0xffffff00 broadcast 10.0.1.255
        ...
        status: active
```
So in this case I have a 24-bit network (the number of one's in the subnet mask) so my address is 10.0.1.23/24, however, if your subnet is a different size then the application will not be able to "listen" on the broadcast address and you'll get the socket error.

It looks like after you changed the address to the correct version, you had a different error, and that should only appear if you have asked for a logger that doesn't exist like `--debug snork` or the library hasn't loaded correctly, which is strange.