JoelBender/bacpypes #393: IP2VLANRouter: Read/Write specific device's objects

ReshmaPatel24, 2021-03-19:
Setup Details:
- BACnet IP2VLANRouter application running on ubuntu system
- Using YABE from windows system
- Both the systems are in same network

Here the objective is to creates multiple device objects(depending upon number of sensors connected to the ubuntu system) and perform read/write operation to each device based on the request received from YABE.

Using IP2VLANRouter application i can create device objects for my sensor devices with multiple analog/binary objects but, i am not sure that how can i decode read or write message coming from YABE is for which device Instance as we are not getting instance ID of device for which request is received.
Decoding the destination address is required, because my application will need to know that for which device instance what request came and then it can be served accordingly.

Attached logs that shows requests coming to the IP2VLANRouter application when selecting device and then it's object from YABE.

One more thing I would like mention here is, when I run the default IP2VLANRouter application it gives error as shown below:

python bacpypes/samples/IP2VLANRouter.py <IP/netmask> <net1> <net2> --rpm --count 5 --debug bacpypes.app
DEBUG:bacpypes.app.Application:__init__ <bacpypes.local.device.LocalDeviceObject object at 0x7f4b176d63d0> None deviceInfoCache=None aseID=None
DEBUG:bacpypes.app.DeviceInfoCache:__init__
DEBUG:bacpypes.app.Application:    - startup fn: <unbound method WhoIsIAmServices.startup>
Traceback (most recent call last):
  File "bacpypes/samples/IP2VLANRouter.py", line 324, in <module>
    main()
  File "bacpypes/samples/IP2VLANRouter.py", line 309, in main
    objectName='Random-1-%d' % (device_instance,),
  File "bacpypes/samples/IP2VLANRouter.py", line 105, in __init__
    self.presentValue = random.random() * 100.0
  File "/usr/local/lib/python2.7/dist-packages/bacpypes/object.py", line 558, in __setattr__
    return prop.WriteProperty(self, value, direct=True)
  File "bacpypes/samples/IP2VLANRouter.py", line 86, in WriteProperty
    super(RandomValueProperty, self).WriteProperty(obj, value, direct=True)
TypeError: super() argument 1 must be type, not classobj

To avoid that error i just commented below line from WriteProperty definition
- super(RandomValueProperty, self).WriteProperty(obj, value, direct=True)
[Read_Log.txt](https://github.com/JoelBender/bacpypes/files/6171146/Read_Log.txt)

JoelBender, 2021-03-20:
This is a result of the `Property` class _not_ being a "new style class", so please change this:
```python
@bacpypes_debugging
class RandomValueProperty(Property):
    ...
```
to this:
```python
@bacpypes_debugging
class RandomValueProperty(Property, object):
    ...
```
Please let me know if this solves it and I'll roll it into the next release.

JoelBender, 2021-03-21:
I have changed the **Property** class to inherit from **object** so the change I posted won't be necessary in the next release.  Please try the **develop** branch.

ReshmaPatel24, 2021-03-22:
I have tried with the latest develop branch and the error that was previoulsy coming is now resolved.

In my previous reply i also asked about how to get the device object instance from the object(Device's Analog Input/Output object) read/write request.

JoelBender, 2021-03-23:
You cannot get the instance number of the device object in a BACnet device using Read Property services since they require the object identifier in the request.  There is no special "wild card" instance number, the special instance number 4194303 is used to mean "uninitialized" for devices that haven't been configured yet.  Send the device a unicast **Who-Is** request message that is unconstrained, no device instance high limit or low limit, and listen for the **I-Am** message coming back from it, that will contain the device object identifier.

Note that on a busy network you will probably also receive **I-Am** messages from other devices, so you need to be careful about filtering through them to pick out the one you want.  Once you have that device object identifier you can use **Read Property** requests read its property list for the device object properties, and the object list for the other objects in the device.

ReshmaPatel24, 2021-03-24:
I think my question is not conveyed properly, please refer to the below details.

- Using IP2VLANRouter application I am creating multiple device objects(as per the sensors connected to the ubuntu system)
- I am using YABE to read/write the device's object(i.e. Analog Input/output). Here IP2VLANRouter application is getting requests for read/write.
- whenever read/write request comes to the application, I want to identify that for which device(sensor) this read or write came.

For example: I have 3 device objects and each device object has 15-20 number of Analog or Binary Input/output objects. now from YABE i am reading the object analog input: 3 of the 2nd device.
Now in IP2VLANRouter application I want to get the device address/objectInstance for which analog input:3 object read request came.
Read request APDU does contain information about objectIdentifier = ('analogInput', 3), but how can I check the device for which this came?

Also, if I have created device with instance ID 1010, and afterwards I want to update the device object instance to any new number then is it possible to update just device Instance ID? or I will need to delete previous device object and then add the new device object with new instance number?

ReshmaPatel24, 2021-03-26:
@JoelBender, is there any update on this? or if you have any questions or want any further information from my side then also please let me know.