JoelBender/bacpypes #395: Using NATRouter.py

DB-CL, 2021-03-26:
Hi,

I'm trying to use NATRouter.py

### Description

I'm in this kind of situation : 

[image]

_(don't worry about the security, this is just an example, we are working with VPN and good firewall configuration on all sides)_

And I'm trying to talk to the BBMD located in a 10.0.0.X network through a NAT.  For this I thought I could use NATRouter.py but I'm really struggling with it.

I'm launching NATRouter.py with this command : 

    NATRouter.py 10.0.0.0/24 47808 1 192.168.0.2 47809 2 --debug

### Problem 1

First of all, I made a correction in the code and I'm not sure it's the right one. Line 70 of the file I changed 

    nat_addr = Address("{}:{}".format(addr1, port2))

to 

    nat_addr = Address("{}:{}".format(addr2, port2))

Because with the original line, the program was not listening on my network 192.168.0.X at all. It only listened on 10.0.0.0:

    udp        0      0 10.0.0.255:47809   0.0.0.0:*               0          76248      5175/python         
    udp        0      0 10.0.0.0:47809     0.0.0.0:*               0          76247      5175/python         
    udp        0      0 10.0.0.255:47808   0.0.0.0:*               0          76246      5175/python         
    udp        0      0 10.0.0.0:47808     0.0.0.0:*               0          76245      5175/python

It was not working at all since there was nothing listening on 192.168.0.2:47809

And after the correction I've made, the netstat -tulpen was more like this

    udp        0      0 192.168.0.2:47809     0.0.0.0:*             0          83849      5884/python         
    udp        0      0 10.0.0.255:47808      0.0.0.0:*             0          83848      5884/python         
    udp        0      0 10.0.0.0:47808        0.0.0.0:*             0          83847      5884/python

That made more sens to me and at least the whois worked that way.

My question is : is it right to make the correction I've made or am I totally wrong about the behaviour of NATRouter ?

### Problem 2

Here is my real problem.

After the correction I've made, the whois was working and I was receiving all the informations from the clients (joy). But I was not able to make a ReadProperty from and external IP :(

Everything looks good, except that the BACnet exchange stops right away.  When we look with Wireshark, we have two BVLC packets followed by a single WhoIs / Iam (Unconfirmed-REQ) with the right device instance. And then nothing. After the device respond to the WhoIs with a Iam, there is no ReadProperty from the client. 

Unfortunately, at the moment I cannot use another client (it's not a BACpypes client). I know this client is working in similar situations because we have some projects with BAS Routers and they do exactly that : creating two bacnet networks and let our client make requests through a NAT, in the exact same way I've explained above.

Any idea what can cause that interruption ?

We tried a working project with a BASRouter and our project with the NATRouter.py and we noticed that on the project with NATRouter.py our client was formatting the BACnet packets differently. In the "Building Automation and Control Network APDU" header, we have a destination network address set to "65535" and the "Control" packet is set to 0x20 in the WhoIs and 0x28 in the Iam. These are the only differences we have between a working project and the NASRouter. I have no idea where that comes from and if it matters or not.

I will try to use a BACPypes client to be able to run more tests but I'm really not sure I will be able to do so.

Thank you for your help

DB-CL, 2021-03-26:
Hmm I think I'm almost there.

I managed to run a BACPypes client. The whois is working well, all BBMD are responding.

The ReadProperty goes further than my first client but it does not complete.

According to the schema I gave in my first post, here is the BACPypes.ini 

    [BACpypes]
    objectName: Betelgeuse
    address: 5.6.7.8/16                      <= this is the VPN network layer
    objectIdentifier: 599
    maxApduLengthAccepted: 1024
    segmentationSupported: segmentedBoth
    vendorIdentifier: 15
    foreignBBMD: 1.2.3.4:27755                <= "public" VPN address, natted to 192.168.0.2
    foreignTTL: 30

And here is my command : 

    read 1:10.0.0.4:47808 2:15 presentValue

Here is the wireshark output, running on the 192.168.0.2 linux station

    37    26.993592    192.168.0.1      192.168.0.2      BACnet-APDU    69    Confirmed-REQ   readProperty[  1] analog-value,15 present-value 
    38    26.999119    10.0.0.2         10.0.0.4         BACnet-APDU    68    Confirmed-REQ   readProperty[  1] analog-value,15 present-value 
    39    27.005261    10.0.0.4         10.0.0.2         BACnet-APDU    75    Complex-ACK     readProperty[  1] analog-value,15 present-value 

We can see the request coming from the NAT (192.168.0.1) and then beeing re-emitted on the BACnet network (10.0.0.X) but the device respond with a Complex-ACK (containing the value) and that's it :( the response does not goes back to my client.

The request is emitted a few times and then the client stops with "noResponse"

It looks like the NATRouter is not forwarding the answer. Am I right ? Any idea Why ?

Thank you very much !

Cyril

ChristianTremblay, 2021-03-26:
You need only one BBMD per subnet.

JoelBender, 2021-03-27:
**Problem 1**

You are correct, I think that bug is in there because the testing I did happened to use the same network for the "inside" network (10.0.0.2/24:47808) and the "NAT side" network (192.168.0.2/24) so that patch is correct, I'll fix the code.

**Problem 2**

The PDUs coming from going from 5.6.7.8:47808 to 1.2.3.4:47809 are expected to come from another NAT router when they land at 192.168.0.2:47809, so they are expected to contain both an SNET/SADR and DNET/DADR.  Unfortunately the Wireshark output does not show the contents of the NPCI.  Turning on debugging for the `bacpypes.netservice.BIPNAT` class will tell you more about what is being sent and received on that port.  It should work, 5.6.7.8:47808 would be the same as FD2 in _Figure J-8_ in Clause J.7.5.

Note also that the NATRouter service presents itself as a BBMD on the inside network, so as Christian says the other BACnet devices should not be a BBMD.  In your case I expect if they _are_ BBMDs then they don't have a BDT or support foreign device registration.

The first parameter you gave NATRouter.py 10.0.0.0/24 is incorrect, it should be 10.0.0.2/24 so I don't think that it is receiving the appropriate unicast PDUs like a confirmed service response.  The IP address 10.0.0.0 is an old-school broadcast address that was used before the concept of all 1's in the host portion of the address was introduced, I don't remember what the motivation was switching from all zeros that was already being used by some socket implementations (like 4.1BSD) to all ones (yes, this is 10BASE5 era, it's been a while).

DB-CL, 2021-03-27:
Hi Cristian, 

Thank you for your message. My scheme is very simplified. Actually behind each BBMD box there is a whole subnet. The BACnet network has been made by a BACnet professional (not me ^^) and I believe it's well done.

Thanks Joel, that was the 10.0.0.2 thing ... As soon as I corrected this, everything started to work properly. No more error in the WhoIs and the ReadProperty is well forwarded. I'm not expert enough to propose you some changes but that could be great to detect an error instead of silently fail the ReadProperty request. No idea how to do that ^^

I proposed a PR with the small error I've found : https://github.com/JoelBender/bacpypes/pull/396

Thanks again

closed 2021-03-27 by DB-CL