JoelBender/bacpypes #404: Reading AddressBinding object from network device returns additional wrong str in objectidentifier

pebauer68, 2021-05-28:
python3.8 with bacpypes 0.18.4
do_read(),args:  ['192.168.x.y/18:47817', 'device:4194xxx', 'manualSlaveAddressBinding']
<Element(objectIdentifier ObjectIdentifier, context=0) instance at 0x06f36b20>
value: ('device', 4194xxx)
<Element(propertyIdentifier PropertyIdentifier, context=1) instance at 0x06f36b50>
value: manualSlaveAddressBinding
<Element(propertyArrayIndex Unsigned, context=2, optional) instance at 0x06f36b80>
value: None
    [0] <bacpypes.basetypes.AddressBinding object at 0x0177C8E0>
        deviceObjectIdentifier = ('analogInput', 0)    # its not analoginput !, just skip this line ?
        deviceAddress
            networkNumber = 0
            macAddress = bytearray(b'\x01')
value:  [<bacpypes.basetypes.AddressBinding object at 0x0177C8E0>]

JoelBender, 2021-05-29:
The content you are seeing is what was encoded in the PDU.  You are correct in that the **deviceObjectIdentifier** sequence element of an **AddressBinding** should be a Device Object identifier, not an Analog Input Object identifier.  I also see that the instance number is zero, which is very unusual and leads me to suspect that you have come across a bug in the device you are talking to.

You will see this in a Wireshark decoding of the incoming response.  To reproduce the encoding if you are looking at a raw hex dump you can build a sample object:
```python
>>> from bacpypes.basetypes import AddressBinding, DeviceAddress
>>> sample = AddressBinding(
...     deviceObjectIdentifier=('analogInput', 0),
...     deviceAddress=DeviceAddress(networkNumber=0, macAddress=b'\x01'),
... )
```
Encode it as a tag list:
```python
>>> from bacpypes.primitivedata import TagList
>>> tag_list = TagList()
>>> sample.encode(tag_list)
>>> tag_list.debug_contents()
        <bacpypes.primitivedata.Tag(objectIdentifier) instance at 0x7fa93b5a8ca0>
            tagClass = 0 application
            tagNumber = 12 objectIdentifier
            tagLVT = 4
            tagData = '00.00.00.00'
        <bacpypes.primitivedata.Tag(unsigned) instance at 0x7fa93b5a8e50>
            tagClass = 0 application
            tagNumber = 2 unsigned
            tagLVT = 1
            tagData = '00'
        <bacpypes.primitivedata.Tag(octetString) instance at 0x7fa93b5a8d60>
            tagClass = 0 application
            tagNumber = 6 octetString
            tagLVT = 1
            tagData = '01'
```
And then encode it as PDU data:
```python
>>> from bacpypes.pdu import PDUData
>>> pdu_data = PDUData()
>>> tag_list.encode(pdu_data)
>>> pdu_data.debug_contents()
    pduData = x'c4.00.00.00.00.21.00.61.01'
```

pebauer68, 2021-05-31:
Hello Joel,
I do not see deviceObjectIdentifier=('analogInput', 0) in Wireshark when I read the Adressbinding from the
device over Network. I Just see ('device',<instance num>) there. 
Could it be a bug in primitivdata.py ? 
Why is every new ObjectIdentifier object initialized with the value ('analogInput', 0) ?

File: primitivdata.py
L: 1714:
**class ObjectIdentifier(Atomic):**

    _app_tag = Tag.objectIdentifierAppTag
    objectTypeClass = ObjectType

    maximum_instance_number = 0x003FFFFF

    def __init__(self, *args):
        self.value = ('analogInput', 0)  **# ?**

ChristianTremblay, 2021-05-31:
Instances are not necessarily starting at 0. You need to read the objectList first.