JoelBender/bacpypes #406: How to run two applications consecutively

MitchCayCRC, 2021-06-09:
I am looking for a way to do this (or am perhaps misunderstanding bacpypes functionality) but it creates communication issues.

I am trying to read values from a device that I am connecting to, but first want to discover what objects are present on this device. Because my device does not support segmentation I cannot create a request for the devices objectList. The idea is to create an application instance which discovers the objects on the device by iterating through the objectList by using indices. Once the full object_list is created there is a call to `stop()` and then a new application instance is created which continuously reads the objects presentValue's and `run()` is called again. I thought this would work but multiple objects will return a `<bacpypes.apdu.AbortPDU(None, )>` instance. However sometimes the second applications ReadPropertyRequest's are successful. Am I missing something?
```
# make a device object
    mstp_args = {
     '_address': int(args.ini.address),
     '_interface':str(args.ini.interface),
     '_max_masters': int(args.ini.max_masters),
     '_baudrate': int(args.ini.baudrate),
     '_maxinfo': int(args.ini.maxinfo),
    }
    this_device = LocalDeviceObject(ini=args.ini, **mstp_args)
    if _debug: _log.debug("    - this_device: %r", this_device)

    # make a discovery application
    discovery_application = DiscoveryApplication(this_device, args.ini.address)
    
    
    

    _log.debug("running discovery application")

    run()

    _log.debug("fini")
    
    # make a read application
    read_application = ReadPointsApplication(args.interval, 
                                             args.mac_address, 
                                             this_device, 
                                             args.ini.address)
    
    _log.debug("running read application")

    run()

    _log.debug("fini")
```

JoelBender, 2021-06-10:
The library is not designed to have more than one application like this, you are probably running into pieces of the first application that aren't disposed of or dereferenced when the second application is built.

MitchCayCRC, 2021-06-10:
Is there no way to do something like this then? I attempted to create a single application with two different types of requests in it; one for doing the discovery of the objects on the device and then another for recurring read requests of those points and seemed to still run into the same issue. My only working solution currently is to create a scheduler that runs the two seperately:
```
def main():
    
    parser = argparse.ArgumentParser()
    
    parser.add_argument('--ini',
                        type=str,
                        help='.ini file to be used (default is bac_client.ini)',
                        nargs='?',
                        default='bac_client.ini')
    
    args = parser.parse_args()
    
    
    os.system(f'python DiscoveryApplication.py --ini {args.ini}')
    os.system(f'python ReadPointListApplication.py --ini {args.ini}')
    
if __name__ == '__main__':
    main()
```

ChristianTremblay, 2021-06-11:
A bacpypes app is a BACnet device on your network. It's not only a client that make read requests on a network like you typically see on client server relationships. 

You don't want a BACnet device to constantly appear and disappear on a network. 

This is why you must consider your app as something that must always run and be accessible for other devices. 

You could technically run other instances in parallel, by using a different port (ex 47809) or defining multiple IP adresses on your NIC, or attaching instances to other NIC... it's a path you can choose. 

I prefer keeping one instance running and "defer" all traffic request to bacpypes run thread. This keep things reliable and work well. 

Speed issues are most of the time created by bad usage of read_properties instead of read property multiple for example... or old devices that could not be faster anyway... 

Bacpypes3 Will bring asyncio capacity to the stack but it's not ready yet.

yujia21, 2025-04-25:
Old issue, but was wondering if this is possible - we want to run two instances in parallel within the same script with multiple IP addresses, but the global variables in run and stop of bacpypes core seem to be blocking this. Running two scripts works since the imports are separated. Do you see any possible way to work around this?

yujia21, 2025-04-25:
Jumping on this old thread - we have a setup where the device running bacpypes is on two networks with different IP addresses and we have bacnet devices on each network. I thought of creating two apps and two devices with different object IDs and different source IPs on one task manager (so one bacpypes.core.run call) to communicate with both bacnet devices, but have yet to be successful. Running two scripts on our device, one for each bacnet device, works fine, but we would prefer to run just one script for other reasons. Since bacpypes.core uses global variables to cache the task manager, is there an issue with having one task manager to handle requests from two local devices?