JoelBender/bacpypes #411: Need example file regarding group object in bacpypes

Rjain18, 2021-07-08:
Hey, 
I would like to implement Group object in my project. I have multiple BACnet devices connected and each device have multiple objects. I saw the object,py file and saw there was provision of GlobalGroupObject and GroupObject. But I was not successful implementing in my code. 
I have search the internet for group object example but haven't found anything useful. It would be helpful if anyone could give me a small example code to implement it.

I have attached a link where they properly explain grouping. I would like to implement the same
https://store.chipkin.com/articles/bacnet-the-group-object

Thank you

JoelBender, 2021-07-08:
Building a GroupObject is relatively simple:
```
    group_1 = GroupObject(
        objectIdentifier=("group", 1),
        objectName="GO1",
        description="Group Object Description",
        listOfGroupMembers=[
            ReadAccessSpecification(
                objectIdentifier=("analogValue", 1),
                listOfPropertyReferences=[
                    PropertyReference(propertyIdentifier="presentValue")
                ],
            )
        ],
    )
```
Trapping the request to read the `presentValue` property of this thing and cycling through the read access specifications is very similar to the `ReadWritePropertyMultipleServices.do_ReadPropertyMultipleRequest()`.

Rjain18, 2021-07-08:
Hey Joel,
First of all thank you for replying so quickly. I appreciate the example code you have sent it was helpful.

Can you also give me an example for GlobalGroupObject. I would want to multiple readable and writable object within the group. 

Thank you

JoelBender, 2021-07-09:
Here is an example:
```python
    global_group_1 = GlobalGroupObject(
        objectIdentifier=("globalGroup", 1),
        objectName="GGO1",
        description="Global Group Object Description",
        groupMembers=[
            DeviceObjectPropertyReference(
                objectIdentifier=("analogValue", 1),
                propertyIdentifier="presentValue",
            ),
            DeviceObjectPropertyReference(
                deviceIdentifier=("device", 699),
                objectIdentifier=("binaryValue", 2),
                propertyIdentifier="presentValue",
            ),
        ],
        groupMemberNames=["my avo 1", "other bvo 2"],
    )
```
Notice that rather than being patterned after the **Read Property Multiple** service, the members are simply listed as references.  The expectation is that the process for building the appropriate request for each of the "external" references will be something separate that can organize the common requests together and perhaps take advantage of Read Property Multiple if it is available from the _other_ device(s).  If you are writing the code for this, you also might want to consider making parallel requests to other devices (like if your group is all of the discharge air flows from a bank of fans that each have their own BACnet VFD, or all of the supply air temperatures from a collection of VAV terminal units).  This is not so easy to do in BACpypes right now.

This is a perfect example where a BACnet server briefly turns into a client of some other device during the processing of a request.

Note that I deliberately listed the `deviceIdentifier` first for the external reference, even though it is the last sequence element of the `DeviceObjectPropertyReference`.  I like to see that up front in the code, the library will of course put it in the right order when it is encoded.

JoelBender, 2021-07-09:
Oh, and I notice that in Clause **20.50.7 Present_Value** of the Global Group Object, there is the added twist that the values are stored locally and may be updated via COV notifications, polling, or a combination of the two.

Rjain18, 2021-07-10:
Hey Joel, 

Let me explain to you what I am looking for in my application, so that you may show me a way on how I should proceed.
I would like to group my objects in YABE. So there will be different types of objects binary, analog and multistate. All the present values should be readable and writable. 

[image]
As you can see in the above image I could manage to make a groupObject (thanks to you) and also add some analog and binary presentValue properties. But I would like to add the whole binary and analog object.

I have to use grouping cause in my application I have to create thousands of BACnet nodes and I would like to group those nodes so that it would be user-friendly. 

For example:- 
Hotel_Room_1_group
          1. AnalogValueObject
          2. AnalogInputObject
          3. BinaryValueObject
          4. BinaryOutputObject
          5. MultiStateValueObject

Any help from you would be appritiated

Rjain18, 2021-07-10:
I would also like to add that I am using BAC0 and BACpypes library. 

`AV_list = ReadAccessSpecification(objectIdentifier=("analogValue",1),
                                  listOfPropertyReferences=[PropertyReference(propertyIdentifier="presentValue")]                          
                                 )

AV_presentValue = ReadAccessResult(objectIdentifier=("analogValue",2),
                                   listOfResults=[
                                       ReadAccessResultElement(propertyIdentifier="presentValue",
                                                               readResult= ReadAccessResultElementChoice(propertyValue = Any() )
                                                               )
                                                   ]
                                   )

group_1 = GroupObject(
       objectIdentifier=("group", 1),
       objectName="GO1",
       description="Group Object Description",
       listOfGroupMembers=[AV_list],
       presentValue=[AV_presentValue]
   )

bacnet.this_application.add_object(group_1)`

I tried to add present value in the group object. But I still didn't get any results.

Rjain18, 2021-07-13:
Hey @JoelBender,
I am really stuck with the above problem. Without grouping the object It would be really difficult for a user to use BACnet with hundreds of objects. I just need to make a group of objects so that a user can easily interface with the system. 

Any help from your side would be appreciated.
Thank you

ChristianTremblay, 2021-07-28:
The idea of using groups is probably a good one in the context of a device having lots of objects. But I must admit that I never saw any device implementing this in the last 20 years. 
Some big devices will just not care and dump 1000's of objects, with bad names and no description.
Other will use some kind of naming convention to sort the different objects.

Rjain18, 2021-08-08:
I will try to look into it. 

Thank you for this amazing library @ChristianTremblay and @JoelBender

closed 2021-08-08 by Rjain18