JoelBender/bacpypes #412: Writing Schedule Present Value

angelenriquecano, 2021-07-12:
Hi:
I need help to write the Schedule's present value property on a device. I am using `ReadWriteProperty.py`. When I make a read I get this response: ` read 1:0x0109 schedule:14 presentValue
DEBUG:__main__.ReadWritePropertyConsoleCmd:do_read ['1:0x0109', 'schedule:14', 'presentValue']
DEBUG:__main__.ReadWritePropertyConsoleCmd:    - request: <bacpypes.apdu.ReadPropertyRequest(12) instance at 0x7fca45c85280>
    <bacpypes.apdu.ReadPropertyRequest(12) instance at 0x7fca45c85280>
        pduDestination = <Address 1:0x0109>
        pduExpectingReply = 1
        pduNetworkPriority = 0
        apduType = 0
        apduService = 12
        objectIdentifier = ('schedule', 14)
        propertyIdentifier = 'presentValue'
        pduData = x''
DEBUG:__main__.ReadWritePropertyConsoleCmd:    - iocb: <bacpypes.iocb.IOCB(1) instance at 0x7fca45cbafd0>
    <bacpypes.iocb.IOCB(1) instance at 0x7fca45cbafd0>
        args = (<bacpypes.apdu.ReadPropertyRequest(12) instance at 0x7fca45c85280>,)
        kwargs = {}
        ioState = 0
        ioComplete = <threading.Event object at 0x7fca45cba760>
        ioCallback = []
        ioPriority = 0
DEBUG:__main__.ReadWritePropertyConsoleCmd:    - datatype: <class 'bacpypes.constructeddata.AnyAtomic'>
DEBUG:__main__.ReadWritePropertyConsoleCmd:    - value: <bacpypes.primitivedata.Enumerated object at 0x7fca45c15190>
Enumerated(1)`
However, I do not know how to send a write's request using the correct datatype.  I have tested using Integer, Unsigned, Boolean and so on, but I got the same response:  `DEBUG:__main__.ReadWritePropertyConsoleCmd:    - iocb: <bacpypes.iocb.IOCB(6) instance at 0x7fca45c06910>
    <bacpypes.iocb.IOCB(6) instance at 0x7fca45c06910>
        args = (<bacpypes.apdu.WritePropertyRequest(15) instance at 0x7fca45c0ad30>,)
        kwargs = {}
        ioState = 0
        ioComplete = <threading.Event object at 0x7fca45c06fa0>
        ioCallback = []
        ioPriority = 0
property: datatypeNotSupported
`
I'll thank any advice and help.
Thanks in advance.
Best Regards, Angel

JoelBender, 2021-07-12:
The value part of a `TimeValue` is any atomic, so the best option (apart from some kind of documentation for the device :smile:) is to read the `weeklySchedule` property, dig through the result to see what's there, and match it.

angelenriquecano, 2021-07-12:
Hi Joel:
Thanks for your rapid answer. 
[image]
I attached a screenshot of the weekly schedule. As you can see, the value has been declared as `Enumerated` and I really have not found a way to modify it.
Thank you again for the support!

JoelBender, 2021-07-12:
Your last best hope is `Enumerated(1)`.  There is no way to know using BACnet to understand what that value means or what it might do... 0=continue normally, 1=crash and burn, 2=ask for a refund, etc.  If you can peer into the rest of the configuration of device and see how the value is being used it would help.

angelenriquecano, 2021-07-13:
Joel, the device uses two states: 0=inactive and 1=active. I sincerely think that a boolean type would have solved the problem, but I do not have access to change the device settings. Best regards!

JoelBender, 2021-07-13:
Ah, good work, it's like the present value (standard: **Present_Value**, BACpypes: `presentValue`) of a Binary Input/Output/Value Object, a (standard ASN.1: **BACnetBinaryPV**, BACpypes: `BinaryPV`).

closed 2021-07-13 by JoelBender

angelenriquecano, 2021-07-13:
Joel, I solved the issue. I modified your code to create an `Enumerated` value.
` datatype = {
                    'b': Boolean,
                    'u': lambda x: Unsigned(int(x)),
                    'i': lambda x: Integer(int(x)),
                    'r': lambda x: Real(float(x)),
                    'd': lambda x: Double(float(x)),
                    'o': OctetString,
                    'c': CharacterString,
                    'bs': BitString,
                    'date': Date,
                    'time': Time,
                    'id': ObjectIdentifier,
                    'e':Enumerated
                    }[dtype]
                if _debug: ReadWritePropertyConsoleCmd._debug("    - datatype: %r", datatype)

                if datatype is Enumerated:
                    value = datatype(int(dvalue))
                    value.enumerations = {0: 0, 1: 1}                    
                else: 
                    value = datatype(dvalue)`
Thank you for the support.
Best regards!
Angel