JoelBender/bacpypes #413: Add COVSubscription with infinite lifetime fails

bobreim, 2021-07-20:
In my program, I use the BACnetApplication in combination with the ChangeOfValueServices.
At startup, I add subscriptions for broadcasting (address = '*').
When lifetime > 0, it works fine. 
However, it gives the following exception when the lifetime is put to 0:

ERROR:bacpypes.core.run:an error has occurred: list.remove(x): x not in list
Traceback (most recent call last):
  File "/home/bob/ChainlessPiHub/venv/lib/python3.7/site-packages/bacpypes/core.py", line 134, in run
    taskManager.process_task(task)
  File "/home/bob/ChainlessPiHub/venv/lib/python3.7/site-packages/bacpypes/task.py", line 376, in process_task
    task.process_task()
  File "/home/bob/ChainlessPiHub/venv/lib/python3.7/site-packages/bacpypes/service/cov.py", line 137, in process_task
    self.cancel_subscription()
  File "/home/bob/ChainlessPiHub/venv/lib/python3.7/site-packages/bacpypes/service/cov.py", line 117, in cancel_subscription
    self.obj_ref._app.cancel_subscription(self)
  File "/home/bob/ChainlessPiHub/venv/lib/python3.7/site-packages/bacpypes/service/cov.py", line 612, in cancel_subscription
    cov_detection.cancel_subscription(cov)
  File "/home/bob/ChainlessPiHub/venv/lib/python3.7/site-packages/bacpypes/service/cov.py", line 184, in cancel_subscription
    self.cov_subscriptions.remove(cov)
  File "/home/bob/ChainlessPiHub/venv/lib/python3.7/site-packages/bacpypes/service/cov.py", line 49, in remove
    self.cov_subscriptions.remove(cov)
ValueError: list.remove(x): x not in list

JoelBender, 2021-07-22:
Yep, looks like this code:
```python
        # if lifetime is none, consider permanent subscription (0)        
        self.lifetime = 0 if lifetime is None else lifetime
        self.install_task(delta=self.lifetime)
```
...should not install itself as a task when the lifetime is None/0.  Thanks for the report!

skewty, 2022-05-02:
bump.  was this fixed?  can this issue be closed?

JoelBender, 2022-05-03:
Yes, this was fixed in [this commit](https://github.com/JoelBender/bacpypes/commit/967782c3202831b20d2ad5772c40bf99b0607eda).  I left this open until I made some specific tests for it, but that is probably not going to happen for this version of BACpypes, so I'm closing the issue.

closed 2022-05-03 by JoelBender