JoelBender/bacpypes #414: 'objectName' is not writable through 'WriteProperty'

MitchCayCRC, 2021-07-21:
I created a new writeable analog value object class:
```
@register_object_type
class WritableAnalogValueObject(AnalogValueObject):
    properties = [WritableProperty('presentValue', Real),
                  WritableProperty('objectName', CharacterString, optional=False),
                  WritableProperty('units', EngineeringUnits)]
```

Then later in the script when trying to use WriteProperty like: `ip_application.objectIdentifier[self.obj_id].WriteProperty(self.prop_id, value)` 
(where self.prop_id was 'objectName') returns the error:
 `ExecutionError(errorClass='property', errorCode='writeAccessDenied')`. 
Using WriteProperty works fine for both the presentValue and units. also `ip_application.objectIdentifier[self.obj_id].objectName = value` 
works fine.

JoelBender, 2021-07-22:
The simple approach in [this gist](https://gist.github.com/JoelBender/e7de20c9b207bc3988e38b6d289c5a1b) seems to work fine, so there must be something else going on.  The "direct" assignment `obj.objectName = "snork"` bypasses all of the validation checks so it will always work, even when giving it the wrong type of value, most likely leading to more problems later.

_Warning_: there is no feedback from the object to the application that says that its name or identifier has changed.  The application keeps a pair of dictionaries to quickly find an object, so even if the object name is successfully changed the application will not know and most likely lead to more problems later.

Changing an object name or identifier during the lifetime of the object was never anticipated.

MitchCayCRC, 2021-07-22:
So if a user were to change an object name through something like yabe that change wont be recognized in the object?

ChristianTremblay, 2021-07-22:
Object name are supposed to be unique in a device
And are static in all the controllers I saw.

I would consider dynamically changing an object name a bad thing.

MitchCayCRC, 2021-07-26:
Unfortunately a large client of ours requested this functionality some time ago and our devices already support it. I am also not seeing anything within the BACnet standard that specifically prohibits this (I'm looking at ANSI/ASHRAE Standard 135-2020). 

On the bright side after confirming that the simplified case in Joel's gist worked I went back and attempted to use WriteProperty and it seems to be working just fine. I am only troubled that I don't know what was going wrong earlier... 

In regards to the _warning_ Joel brought up, was that only a warning for if I were to have tried the "direct" assignment or is that true even if I am using WriteProperty?

JoelBender, 2021-07-26:
It would be for both, "changing the object name" should be just that, internal within the application or from an external request.  I can add that feature as a mix-in class like `CurrentPropertyListMixIn`, tagging this as an enhancement.

JoelBender, 2021-07-28:
This enhancement has been added to the **414-objectName-is-not-writable** branch, along with a sample application.  Please test this with your application and let me know how it goes.