JoelBender/bacpypes #426: Creating an object on a remote device

mosterme1003, 2021-09-13:
Hi there! Is it somehow possible to create a bacnet object on a device in the field remotely? The way I understand it the object can be created using the (Un)confirmedPrivateTransfer service but I don't understand what service parameters are needed in this case. Can you help me with that?

JoelBender, 2021-09-13:
Yes, there is a `CreateObject` service and it is relatively simple, all the parameters are the properties of the object that you would like to create, starting with the required ones and perhaps some optional ones as well.  Devices that provide this _optional_ service not only validate that the properties values are correct for the type, but also that if there are optional properties that appear together, they are also provided.

You can create the request like this:
```python
from bacpypes.primitivedata import CharacterString
from bacpypes.constructeddata import Any
from bacpypes.basetypes import PropertyValue
from bacpypes.apdu import CreateObjectRequest, CreateObjectRequestObjectSpecifier

cor = CreateObjectRequest(
    objectSpecifier=CreateObjectRequestObjectSpecifier(objectType="analogValue"),
    listOfInitialValues=[
        PropertyValue(
            propertyIdentifier="objectName", value=Any(CharacterString("some-new-object"))
        )
    ],
)
cor.debug_contents()
```
but note that BACpypes doesn't support this service as a server, only as a client of some other device.

mosterme1003, 2021-09-14:
It worked! Thank you very much. I already tried something like this before but I think there was a mistake in my code.

closed 2021-09-14 by mosterme1003