JoelBender/bacpypes #435: [QUESTION] How to identity MSTP Devices

WeiTangLau, 2021-11-02:
I am looking for docs regarding identifying MSTP Devices hidden behind an IP Device (could be a router), but I couldn't find any. Furthermore, I found an [extension](https://github.com/riptideio/misty) to add MSTP support. Does this mean that MSTP is not supported by this library? If not, how can I identity the MSTP device?

Thanks in advance!

JoelBender, 2021-11-03:
For BACnet devices that are Revision 17 or later _and_ support the **Network Port Object** then you can find out by looking in the **Network_Type** property (Clause 12.56.8), in BACpypes that would be the `networkType` property name.  The value is one of the enumerations of a **BACnetNetworkType**, in BACpypes that is the `NetworkType` class.  An MS/TP device will probably only have one network port object, so it should be easy to find.  If the IP-to-MS/TP router supports network port objects, you should be able to find it that way as well.

Most of the time a single octet address is probably an MS/TP or ARCNET address and there is no way to distinguish them from each other.  If you send an Initialize-Routing-Table to a router, you will get some port information, but most of the time you won't get anything very useful.

The **misty** project adds MS/TP support for directly connected serial port devices and isn't needed for talking through a router.

WeiTangLau, 2021-11-03:
Hi! Thanks for your response. 
My current setup is MSTP -> Router -> Laptop. I do have a router like the case you mentioned. However, I am actually using a third party software that uses this library. As such, I do not have much flexibility. The developer told me that they actually uses [this](https://bacpypes.readthedocs.io/en/latest/tutorial/addressing.html) under the hood. IS this sufficient or has to do what you mentioned above?

JoelBender, 2021-11-03:
You are correct.  The router is going to have two ports, each will have a unique network number.  Let's assume that the IPv4 network is BACnet network 10 and the MS/TP network has BACnet network 20.  Your laptop will be on network 10 and you will be sending messages to the IP address of the router asking it to forward the message to some device on network 20.

First try running the **WhoIsRouter.py** sample application and enter `wirtn *` which means "send out a local broadcast on this network to all of the routers and have them respond with the list of attached networks." You should see a message from the router saying "I am router to network 20."

Now try running the **WhoIsIAm.py** application and enter `whois *:*` which means "send out a _global_ broadcast message to every BACnet node and have them respond with an I-Am message.  You should see at least one from a device on the MS/TP network, the address will look like `20:13` which in BACpypes means "address 13 on network 20."

All of the rest of you read and write property requests should be sent to `20:13` and the router will take care of shuffling the packets between networks.