JoelBender/bacpypes #439: Python 3.9.9 issue

ELASHALL, 2021-12-11:
i got this error!

  File "bacnet/bacpypes/setup.py", line 26, in <module>
    raise EnvironmentError("unsupported version of Python")
OSError: unsupported version of Python
python -V                                                                                                                                                 
Python 3.9.9
would you please help me how to fix this? should i downgrade my python to older version?

JoelBender, 2021-12-11:
Check out (clone) the project and create a symlink (alias) to the `py34` directory:
```bash
$ ln -s py34 py39
```
And then run `python3 setup.py install`.  There _shouldn't_ be anything that prevents it from working, but it has an "old school" socket event loop (in  the `run()` function in the `core` module) and doesn't use async/await.  The next version, BACpypes3, is asyncio based, Python3.8+.

ELASHALL, 2021-12-12:
the same error still show up
please , how to download BACpypes3?

ELASHALL, 2021-12-12:
it still give the same error
Traceback (most recent call last):
  File "/home/elashall/bacnet/bacpypes/setup.py", line 26, in <module>
    raise EnvironmentError("unsupported version of Python")
OSError: unsupported version of Python
how to download BACpypes3?

JoelBender, 2021-12-12:
Ah!  I forgot that there is a version to directory dictionary in `setup.py` because I'm accustomed to using symlinks.  Edit that and you should be good to go.  BACpypes3 has not been released yet.

ELASHALL, 2021-12-12:
thank you
i add this:
 (3, 9): 'py39',
it works fine thank you sir

closed 2021-12-13 by ELASHALL