JoelBender/bacpypes #440: Unable to deserialize and serialize Address objects to and from strings

yagnasrinath, 2021-12-14:
```
python3
Python 3.9.6 (default, Jun 29 2021, 05:25:02)
[Clang 12.0.5 (clang-1205.0.22.9)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> from bacpypes.pdu import RemoteStation, Address
>>> b=Address("6312:9@10.10.15.15:47827")
route provided but not route aware: '6312:9@10.10.15.15:47827'
>>> b
<Address 6312:9@0x0a0a0f0fbad3>
>>> str(b)
'6312:9@0x0a0a0f0fbad3'
>>> Address('6312:9@0x0a0a0f0fbad3')
route provided but not route aware: '6312:9@0x0a0a0f0fbad3'
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.9/site-packages/bacpypes/pdu.py", line 71, in __init__
    self.decode_address(args[0])
  File "/usr/local/lib/python3.9/site-packages/bacpypes/pdu.py", line 207, in decode_address
    self.addrRoute = Address(int(route_addr))
ValueError: invalid literal for int() with base 10: '0x0a0a0f0fbad3'

```

yagnasrinath, 2021-12-14:
I have a couple of fixes for the same, But I am not sure of the consequences of these fixes. I can raise a PR if you would like with one of these fixes:
**Fix 1**:
```
206,208c206,212
<                 if route_addr:
<                     self.addrRoute = Address(int(route_addr))
<                     if _debug: Address._debug("    - addrRoute: %r", self.addrRoute)
---
>                 if re.match(r"^0x([0-9A-Fa-f][0-9A-Fa-f])+$",route_addr):
>                         if _debug: Address._debug("    - modern hex string")
>
>                         self.addrRoute = Address(xtob(route_addr[2:]))
>                     else:
>                         self.addrRoute = Address(int(route_addr))
>                         if _debug: Address._debug("    - addrRoute: %r", self.addrRoute)
```

**Fix 2**:
```
207,208c207
<                     self.addrRoute = Address(int(route_addr))
<                     if _debug: Address._debug("    - addrRoute: %r", self.addrRoute)
---
>                     self.addrRoute = Address(route_addr)
```

**Fix 3**: 
```
405c405
<                 if (len(self.addrAddr) == 6) and (port >= 47808) and (port <= 47823):
---
>                 if (len(self.addrAddr) == 6):
421c421
<                 if (len(self.addrAddr) == 6) and (port >= 47808) and (port <= 47823):
---
>                 if (len(self.addrAddr) == 6):
```

JoelBender, 2021-12-14:
Check out [this commit](https://github.com/JoelBender/bacpypes/commit/f851e532e9dcceb5477792ca7d688a32fbe79a98), it should be working correctly.

yagnasrinath, 2021-12-14:
Thanks it worked