JoelBender/bacpypes #442: WriteProperty() vs. direct assignment (COVserver.py)

villeilkkala, 2022-02-03:
Studying COVserver.py example, I can see two ways of assigning a new value to the `presentValue` property.

In the `do_write()` function the assignment of the new value is handled with the `WriteProperty` method: `obj.WriteProperty(property_name, value)`

in the `TestAnalogValueThread()` class the assignment of the new value is done directly: `test_av.presentValue = next_value`.

Looking at the implementation, the `WriteProperty()` seems to have a lot of error handling. Replacing the direct assignment from the `TestAnalogValueThread()` class with `WriteProperty()` also works the same.

In what scenarios should I use `WriteProperty()` and when do direct assignments?

JoelBender, 2022-02-03:
The difference is to test what happens when the property value is changed internally (like `x.y = 3` in some code someplace) vs what happens when the change comes from an external source via WriteProperty request.  You can do one or the other, they should behave similarly.  The extra error checking is avoided in the "direct assignment" case, assuming the developer using the library is setting property values to their correct type and value.

villeilkkala, 2022-02-22:
thanks, this is clear.

closed 2022-02-22 by villeilkkala