JoelBender/bacpypes #443: Virtual devices are very slow to read

barrashi, 2022-02-09:
using the samples, if i use the mini_device.py the speed is fine

if i use the IPTOVLANRouter.py
it take more than a second to list properties

Anyway i could speed this up?

Thanks

JoelBender, 2022-02-09:
There is very little difference between the two applications, what are you using to "list properties"?  My guess is that it's the difference between reading values in large batches using Read Property Multiple and reading them one at a time using Read Property.

barrashi, 2022-02-10:
Thank you very much for your response
and contribution!

I am using YABE to read the data.

currently i am just running two samples as they are without changing
anything.

the virtual devices are very slow to read(IP2VLANRouter.py)

and the single instance (mini_device.p)
works fast!.

also, i am unable to get the COV to work with the virtual devices on the
sample, or in any other amature attempts i have done.

if you can take the time and help me here that would be awesome.

thanks.