JoelBender/bacpypes #448: TypeError: unsupported operand type(s) for -: 'float' and 'NoneType'` in cov.py

villeilkkala, 2022-03-02:
I overwrite a `presentValue` of an analogValueObject in my COV application. I get the following error after a thousand or so overwrites:

`Traceback (most recent call last):
  File "/usr/local/lib/python3.9/threading.py", line 973, in _bootstrap_inner
    self.run()
  File "/code/./main.py", line 230, in run
    bacnet_object.WriteProperty("presentValue", value)
  File "/usr/local/lib/python3.9/site-packages/bacpypes/object.py", line 620, in WriteProperty
    return prop.WriteProperty(self, value, arrayIndex, priority, direct)
  File "/usr/local/lib/python3.9/site-packages/bacpypes/object.py", line 363, in WriteProperty
    fn(old_value, value)
  File "/usr/local/lib/python3.9/site-packages/bacpypes/service/detect.py", line 45, in property_change
    trigger = self.filter(old_value, new_value)
  File "/usr/local/lib/python3.9/site-packages/bacpypes/service/cov.py", line 317, in present_value_filter
    value_changed = (new_value <= (self.previous_reported_value - self.obj.covIncrement)) \
TypeError: unsupported operand type(s) for -: 'float' and 'NoneType'`

When I restart the service, a lot of messages come in in quick succession, so it happens almost immediately. The rate of overwrites slows down to 6 per minute after a while.

I don't see anything from the written values that would indicate anything:

`
Writing 6.43 to <__main__.WritableAnalogValueObject object at 0x7f41eadc3130>
Writing 5.31 to <__main__.WritableAnalogValueObject object at 0x7f41eadc3130>
Writing 2.89 to <__main__.WritableAnalogValueObject object at 0x7f41eadc3130>
Writing 7.14 to <__main__.WritableAnalogValueObject object at 0x7f41eadc3130>
Writing 3.94 to <__main__.WritableAnalogValueObject object at 0x7f41eadc3130>
Writing 3.18 to <__main__.WritableAnalogValueObject object at 0x7f41eadc3130>
Writing 0.07 to <__main__.WritableAnalogValueObject object at 0x7f41eadc3130>
Writing 8.87 to <__main__.WritableAnalogValueObject object at 0x7f41eadc3130>
Writing 4.87 to <__main__.WritableAnalogValueObject object at 0x7f41eadc3130>
Writing 1.87 to <__main__.WritableAnalogValueObject object at 0x7f41eadc3130>
Writing 1.78 to <__main__.WritableAnalogValueObject object at 0x7f41eadc3130>
Writing 7.13 to <__main__.WritableAnalogValueObject object at 0x7f41eadc3130>
Writing 4.85 to <__main__.WritableAnalogValueObject object at 0x7f41eadc3130>
unsupported operand type(s) for -: 'float' and 'NoneType'
Writing 4.84 to <__main__.WritableAnalogValueObject object at 0x7f41eadc3130>
unsupported operand type(s) for -: 'float' and 'NoneType'
Writing 7.27 to <__main__.WritableAnalogValueObject object at 0x7f41eadc3130>
unsupported operand type(s) for -: 'float' and 'NoneType'
Writing 4.75 to <__main__.WritableAnalogValueObject object at 0x7f41eadc3130>
unsupported operand type(s) for -: 'float' and 'NoneType'
Writing 6.3 to <__main__.WritableAnalogValueObject object at 0x7f41eadc3130>
unsupported operand type(s) for -: 'float' and 'NoneType'
Writing 3.56 to <__main__.WritableAnalogValueObject object at 0x7f41eadc3130>
unsupported operand type(s) for -: 'float' and 'NoneType'
Writing 6.95 to <__main__.WritableAnalogValueObject object at 0x7f41eadc3130>
unsupported operand type(s) for -: 'float' and 'NoneType'
Writing 6.94 to <__main__.WritableAnalogValueObject object at 0x7f41eadc3130>`

This has been tested with covIncrement=1.0 and without specifying one.

JoelBender, 2022-03-02:
It would seem there is some tracking of what was previously reported that is getting cleared.  Is this code using threads?  I'm going to try and reproduce this error.

villeilkkala, 2022-03-02:
Yes, this is running in a separate thread.