JoelBender/bacpypes #453: Client-only Applications example

vakker, 2022-04-11:
Hi,

The documentation says that it's possible to make client-only applications since 0.15.0 (see [here](https://bacpypes.readthedocs.io/en/latest/migration/migration001.html#client-only-applications)).
Could you provide a simple example for WhoIs and ReadProperty that uses the client-only version?
I tried to modify `BIPSimpleApplication` to remove the device and the address bindings, but that doesn't seem to work.

JoelBender, 2022-04-13:
That code was quite a while ago, so I believe that what will be necessary is to create an application stack that still has a device object for properties like segmentation support and such, will still need an address to know how to send and receive traffic, but does not include WhoIsIAmServices and ReadWritePropertyServices, so it will not respond to those requests.

JoelBender, 2022-04-14:
[This gist](https://gist.github.com/JoelBender/e8413dfb466789a7b97dd095e0c2d172) is a starting point.

closed 2022-04-14 by JoelBender

vakker, 2022-04-14:
Thanks! Is there a way not bind the IP address, so multiple apps can be instantiated?

JoelBender, 2022-04-14:
As long as you don't need any broadcast services, which means you won't use Who-Is or Who-Is-Router-To-Network to discover devices and the network topology, you can use ephemeral ports by specifying "12.23.34.45:0" (or something like "eth0:0") as the address and the socket layer will pick an unused port.