JoelBender/bacpypes #454: Activating and deactivating BBMD functionality

mosterme1003, 2022-04-13:
Hey there! I tested your added function (#451) for writing the BDT to an already working BBMD and it works perfectly fine. Now I wonder if it's possible to activate and deactivate the BBMD functionality of a device. When I try to write the BDT on a device that is not yet configured as a BBMD I get back a NAK. And I get the same result when I try to write an empty BDT to the BBMD in order to deactivate the functionality. Is there a way activate and deactivate it? Is there another BACnet service you need to use or is it maybe something proprietary? I kinda worked my way through annex J and couldn't find an answer. Thank you for your help!

JoelBender, 2022-04-13:
There isn't anything in Annex J that will help, however, and you _might_ think that **DeviceCommunicationControl** would be part of the asnwer, but that is at the application layer and not the virtual link layer.

Check out _Clause 12.56 Network Port Object Type_.  I do not have much experience with this beast because our devices don't support it or it hasn't been something high enough on my to-do list.  In theory, if the device has a NetworkPortObject and the Out_Of_Service property is writable, then you can...

1. change Out_Of_Service to True, the changes will be "pending"
2. write a RESTART_PORT to the Command property
3. check for a Result(+)

But because the NPO has multiple protocol layer content mashed in together via Network Port Hierarchies (Clause 12.56.10.1) that may disable more than just the BBMD functionality.  It's more likely that you want to 

1. change the BBMD_Broadcast_Distribution_Table property to an empty table.
2. change BBMD_Accept_FD_Registrations to `False`
3. change BBMD_Foreign_Device_Table to an empty table, and finally
4. send a ReinitializeDevice (Clause 16.4) service request with ACTIVATE_CHANGES.

Good luck!  BACpypes will not support this kind of thing until BACpypes3 has escaped into the wild.

mosterme1003, 2022-07-20:
Unfortunately it seems that at the current state and with the BACnet revisions that our devices use I will not be able to activate or deactivate the BBMD functionality. The only thing I can do is to empty the BDT like you said. But thats okay. Thank you for your help and your awesome work for the BACnet community.

closed 2022-07-20 by mosterme1003