JoelBender/bacpypes #458: sending object value packet without ack packet

pooyahmt, 2022-05-26:
@JoelBender Hi
first of all thanks for great package and efforts you have done. you really did a great job.
I'm new to bacnet and thanks to your package I succeeded to create objects and communicate send and receive serial data through bacnet and communicate with bacnet server. I wanted to ask is it possible that we send readproperty packet to the specific destination address without acknowledgment from destination device ?(for example streaming packets containing property without running yabe)
[image]

I need 69 complex-ack packet to be sent with a timer without 59 confirmed-request from destination.

JoelBender, 2022-05-27:
If you are the "client" sending requests, there is no mechanism to ask the "server" don't send back an acknowledgement.  You can, however, form a packet like you would at the application layer which can be a request or an acknowledgement (or error or abort) and then just send it, ignoring a response if you get one.  Run simple read property application with debugging turned on and look at the APDU as it goes downstream from the application into the network service access point.  You will want to replace the segmentation state machine and the application above it with another subclass of `Client` that forms the request and calls `request()` sending it down the stack and include a blank `confirmation()` method that throws out what it gets upstream.

Note that you will need to manage the invoke identifier yourself and increment it between each request (or maybe you want to repeat it for some reason).  It's also quite easy to flood a device, essentially a denial-of-service attack, and cause all kinds of problems.

pooyahmt, 2022-07-10:
Thanks a lot for your useful repository.

closed 2022-07-10 by pooyahmt