JoelBender/bacpypes #460: [QUESTION] How to give non-IP devices an address.

brandonbeattie22, 2022-05-31:
I am very new to BACNet and trying to familiarize myself with its concepts as I have been tossed into a situation involving BACNet and this library. I am very sorry if these types of questions are not allowed.

I have something very similar to IP2VLANRouter.py running on a raspberry pi, but with some radio devices with their various readings instead of random devices as in the sample. In order to integrate this setup into our building, we have sent some units over for testing and were told that due to how the devices show up, they aren't able to integrate. They show up in YABE but apparently with a MAC address of 1, 2, 3, or 4 instead of an actual IP address. I am struggling to find out how to update this code to bring it up to their standards. I am able to use YABE to read the values perfectly fine, so I am not 100% on what the actual issue is.

JoelBender, 2022-05-31:
### It's not you!

I understand that your radio devices a "virtual devices" running on a "virtual network" inside the router application, which is perfectly fine, and being able to use YABE (along with some Wireshark captures of the BACnet traffic being correct) reinforces that conclusion.  BACnet addresses have two pieces, a network portion (unsigned 16-bit integer) and a MAC address that can vary in size depending on the network (so little networks have little single-octet numbers, Ethernet and IPv4 six octets, etc).

In order for them to talk to your virtual devices, they have to know the IP address of your router and the virtual network number and the MAC address your devices are providing, and from your description it sounds like they can't handle BACnet devices that are on non-IPv4 networks, which is quite seriously broken and you can be justifiably irritated.

There are two solutions:

1. Configure the pi with multiple IP addresses, a.k.a., single link multi-homing.
2. Rebuild your device as a single server and map each of the properties of your transceivers to a range of object identifiers; where you had virtual device 2 and analog value 12, map that to 20012, then virtual device 3 and the same analog value 12, map that to 30012, etc.

Option 1 is non-trivial to get right, _especially_ because all of the virtual devices will be on the same IPv4 network and share a broadcast listening socket (i.e., x.y.z.255) and the other vendor may _still_ complain because they don't want to deal with all of the devices individually, just as an aggregated pool of stuff they can cherry pick from.

Your only realistic option is 2.

brandonbeattie22, 2022-05-31:
Thank you for your super speedy and very helpful answer! That makes a lot of sense and seems not too difficult to implement using your library. I'll close this issue.

closed 2022-05-31 by brandonbeattie22