JoelBender/bacpypes #465: Register a foreign device on multiple BBMD

mosterme1003, 2022-07-20:
Hey there! I want to connect a BACnet device with multiple subnets by registering it as a foreign device on the given BBMDs. My first approach was creating multiple BACnet devices using `BIPForeignApplication` and different ports. Now I wonder if it was possible to create just one device and register it on multiple foreign BBMDs. I tried to modify the `BIPForeign` class but I got issues with the layer bindings. Could you help me with this problem? Would it be complex to add this functionality?

JoelBender, 2022-07-20:
There is one set of problems at the virtual link layer, the `BIPForeign` class expects to be associated with a single BBMD and maintain its registration state.  If you have multiple instances of that using the same socket layer, you'll have to write an additional layer `X` that merges the downstream traffic and then splits out the upstream traffic based on the source address.  From the `BIPForeign` class instances `F1` and `F2` you have the foundation of another stack, where Nx is a network layer and Ax is an application layer.
```mermaid
graph TD;
    A1-->N1;
    A2-->N2;
    N1-->F1;
    N2-->F2;
    F1-->X;
    F2-->X;
    X-->Y;
```
You could have one application and network layer, then route the packets, assuming that you don't have the same network number used in both networks you are connecting to, with the additional downside that you will be routing broadcast messages between the two networks which could be really surprising:
```mermaid
graph TD;
    A-->N;
    N-->F1;
    N-->F2;
    F1-->X;
    F2-->X;
    X-->Y;
```
I would expect that you want to use the same device "profile" (instance number, device name, etc) and "appear" on each network like each one is a "site" (BACnet intranet).  For that you need multiple application stacks and manage the requests and responses for each site siloed from each other.

mosterme1003, 2023-05-04:
Thanks for your detailed answer. Since I haven't looked into network layers much before, I guess I'll work around this problem for now by creating multiple BACnet devices under different addresses.

closed 2023-05-04 by mosterme1003