JoelBender/bacpypes #466: How to read a private Object and new property

Darrendong, 2022-08-05:
Hello.
I want to write the code to read a new Object and new property, Object number 129, property number 547. This is defined by the supplier. I use Yabe read the data as below. When I try to read the data from ReadPropertyAny.py,  use read 1 129:0 547, the datatype is not supported.

May I know what should I do? Add one new object in object.py? or any other fast ways?

thank you

[image]

Darrendong, 2022-08-05:
this is the error

[image]

JoelBender, 2022-08-05:
Is that an old version of **ReadPropertyAny.py**?  That `cast_out(Unsigned)` call is expecting and Unsigned value and the [current version of the code](https://github.com/JoelBender/bacpypes/blob/master/samples/ReadPropertyAny.py#L94) looks for an application tag and then the data type.  If it's context encoded (like the vendor specific property value is a choice or some other structure with context encoded content) then there's nothing a generic application can do.

Darrendong, 2022-08-08:
Thank you for your reply.
I use the misty, which is for Bacnet MS/TP, it is using bacpypes.
[image]
The version is as below:
 pip3 install misty
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: misty in ./.local/lib/python3.6/site-packages (0.0.13)
Requirement already satisfied: bacpypes>=0.18.0 in ./.local/lib/python3.6/site-packages (from misty) (0.18.6)
Requirement already satisfied: six>=1.15.0 in ./.local/lib/python3.6/site-packages (from misty) (1.16.0)

 and the device returns a string of the Object number 129, property number 547.

What should I do?

JoelBender, 2022-08-08:
From the standard:

> Enumerated values 512-4194303 may be used by others subject to the procedures and constraints described in Clause 23.

Property identifier 547 is vendor specific and you'll need guidance from the vendor on how to interpret the value.  If you attach a debugging logger where you are getting the response (calling `debug_contents()` on the APDU or pieces inside it) you can sometimes figure out what it is (boolean, integer, etc.) but not what it means.

Darrendong, 2022-08-11:
Thank you, Joel Bender.

It is my problem, I just use your example code of [ReadProperty.py], but not your  [ReadPropertyAny.py]. After I fix this, now it can work.

thank you for your reply.

closed 2022-08-11 by Darrendong