JoelBender/bacpypes #468: Implement WhoIsIAm and "WhoHasIHave services together

JoelBender, 2022-09-14:
Can you please guide me on how to implement "**WhoIsIAm**" and "**WhoHasIHave**" services **together**? 

I have done WhoIsIAm and am through with it and have also implemented multiple devices in one application but now I want to provide WhoHasIHave Service also. So, Can you please help me with that?!

_Originally posted by @SRikinR in https://github.com/JoelBender/bacpypes/issues/189#issuecomment-1243229082_

JoelBender, 2022-09-14:
The **WhoHasIHaveServices** class in the **bacpypes.service.device** module is a "mix-in class" which you can include in your inheritance class list when you subclass an **Application** or one of its subclasses, like **BIPSimpleApplication**.  In the ReadPropertyMultipleServer.py sample application you can see where the [ReadPropertyMultipleApplication](https://github.com/JoelBender/bacpypes/blob/master/samples/ReadPropertyMultipleServer.py#L35) class inherits from BIPSimpleApplication and includes additional services like **ReadWritePropertyMultipleServices** and **DeviceCommunicationControlServices**.

SRikinR, 2022-09-16:
We have already done something similar to this, please look into it and help us further. 

Here we have described how we have created the **Application class** & **LocalDeviceObject**:

1) Make the application  

```
vlan_app = VLANApplication(vlan_device, vlan_address)
> vlan_device = \                                                           
          LocalDeviceObject(                                                                            
              objectName= "ABCD" + str(device_instance), #name update    
              objectIdentifier=('device', device_instance),                            
              maxApduLengthAccepted=1024,                                   
              segmentationSupported='noSegmentation',                       
              vendorIdentifier= <vendor_id>,                                         
              )         

 > vlan_address = Address(2)        
```

2) **Application Class**
```
class VLANApplication(                                                          
    Application,                                                                
    WhoIsIAmServices,                                                           
    WhoHasIHaveServices,                                                        
    ReadWritePropertyServices,                                                        
    ):                                                                          
                                                                                
    def __init__(self, vlan_device, vlan_address, aseID=None):                  
        if _debug: VLANApplication._debug("__init__ %r %r aseID=%r", vlan_device, vlan_address, aseID)
        global args                                                             
                                                                                
        # normal initialization                                                 
        Application.__init__(self, vlan_device, aseID=aseID)                    
        # optional read property multiple                                       
        #if args.rpm:                                                           
        #    self.add_capability(ReadWritePropertyMultipleServices)             
                                                                                
        # include a application decoder                                         
        self.asap = ApplicationServiceAccessPoint()                             
                                                                                
        # pass the device object to the state machine access point so it        
        # can know if it should support segmentation                            
        self.smap = StateMachineAccessPoint(vlan_device)                        
                                                                                
        # the segmentation state machines need access to the same device        
        # information cache as the application                                  
        self.smap.deviceInfoCache = self.deviceInfoCache                        
                                                                                
        # a network service access point will be needed                         
        self.nsap = NetworkServiceAccessPoint()                                 
                                                                                
        # give the NSAP a generic network layer service element                 
        self.nse = NetworkServiceElement()                                      
        bind(self.nse, self.nsap)                                               
                                                                                
        # bind the top layers                                                   
        bind(self, self.asap, self.smap, self.nsap)                             
                                                                                
        # create a vlan node at the assigned address                            
        self.vlan_node = Node(vlan_address)                                     
                                                                                
        # bind the stack to the node, no network number, no addresss            
        self.nsap.bind(self.vlan_node)                                          
                                                                                
    def request(self, apdu):                                                    
        if _debug: VLANApplication._debug("[%s]request %r", self.vlan_node.address, apdu)
        Application.request(self, apdu)                                         
                                                                                
[... 7 lines cut ...]
                                                                                
    def confirmation(self, apdu):                                               
        if _debug: VLANApplication._debug("[%s]confirmation %r", self.vlan_node.address, apdu)
        Application.confirmation(self, apdu)                                    

```

We are able to capture **i-am** packets through Wireshark but are unable to see **i-have** packets. we are using **YABE** as a BACnet application and we do not find anythings that sends a "who-has" request in it.

[... cut ...]
objectIdentifier = ('analogInput', 1),                   #object id
objectName= "Liters of water",                          #object name 
)

but in this case, I am not able to get that for both device id and object id we use the same variable "objectIdentifier"!! and also, for the objectName. 
```

I do not to create more confusions but I have also raise a detailed issue here: [https://github.com/JoelBender/bacpypes/issues/467](url). this might be helpful. 

you might be aware of the correct method so, please provide you valuable suggestion.

SRikinR, 2022-09-18:
Thank you for your help, I think only one thread should be active or else this might create confusion and might be kind of hectic. I request you to close this thread and continue our discussion in the #467 
We have same issue thread going on #467.