JoelBender/bacpypes #469: Scanning BACnet devices from a BACnet controller

mohakbhalla7, 2022-09-15:
Hi @JoelBender 
I'll try my best to explain my use case and the problem I am facing. 

**Use Case:** 
So I have a BACnet setup with me where there is a BACnet IP device connected with a BACnet/IP Controller, and that controller is connected with my RPi module through ethernet. Using BACpypes, I want to write a script in my RPi to run a scan and get the details of those BACnet/IP devices connected with the BACnet controller.

**Error:**
I ran your `WhoIsIAm.py` sample code in my RPi with a `.ini` file with address as `10.14.81.40/24` (The RPi has the eth0 address as `10.14.81.48` and subnet-mask as `255.255.255.0`).
Also, I'm using Python2.7.
This is the error I am getting.

```
Traceback (most recent call last):
  File "whois.py", line 193, in <module>
    main()
  File "whois.py", line 177, in main
    this_application = WhoIsIAmApplication(this_device, args.ini.address)
  File "whois.py", line 41, in __init__
    BIPSimpleApplication.__init__(self, *args)
  File "/home/pi/.local/lib/python2.7/site-packages/bacpypes/app.py", line 535, in __init__
    self.mux = UDPMultiplexer(self.localAddress)
  File "/home/pi/.local/lib/python2.7/site-packages/bacpypes/bvllservice.py", line 97, in __init__
    self.directPort = UDPDirector(self.addrTuple)
  File "/home/pi/.local/lib/python2.7/site-packages/bacpypes/udp.py", line 155, in __init__
    self.bind(address)
  File "/usr/lib/python2.7/asyncore.py", line 342, in bind
    return self.socket.bind(addr)
  File "/usr/lib/python2.7/socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
socket.error: [Errno 99] Cannot assign requested address
```

I read the issue #360 . It had a similar error but I think our uses cases are different. Though I tried the things that you suggested in the issue, but they were of no help.

After all this, I have a question. Is it possible to scan, read or write BACnet/IP devices connected with a BACnet controller using BACpypes?

Any help is appreciated.
Thanks!

JoelBender, 2022-09-15:
The address you provide to the application needs to be its "local" address, the one that the application should "listen to."  When the program runs it will prompt you for the whois command and optionally the destination address for the request.  Update the **address:** field in you INI file to **10.14.81.48/24** and try again.

mohakbhalla7, 2022-09-16:
Tried it already. Didn't work with `10.14.81.48/24`.
This was the error I was getting.

```
Traceback (most recent call last):
  File "whois.py", line 193, in <module>
    main()
  File "whois.py", line 177, in main
    this_application = WhoIsIAmApplication(this_device, args.ini.address)
  File "whois.py", line 41, in __init__
    BIPSimpleApplication.__init__(self, *args)
  File "/home/pi/.local/lib/python2.7/site-packages/bacpypes/app.py", line 535, in __init__
    self.mux = UDPMultiplexer(self.localAddress)
  File "/home/pi/.local/lib/python2.7/site-packages/bacpypes/bvllservice.py", line 97, in __init__
    self.directPort = UDPDirector(self.addrTuple)
  File "/home/pi/.local/lib/python2.7/site-packages/bacpypes/udp.py", line 155, in __init__
    self.bind(address)
  File "/usr/lib/python2.7/asyncore.py", line 342, in bind
    return self.socket.bind(addr)
  File "/usr/lib/python2.7/socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
socket.error: [Errno 98] Address already in use
```

JoelBender, 2022-09-17:
If the address is already in use then there is another application running that is already "bound" to the port.  Before you run the application, try something like `sudo netstat -unlp` or `sudo ss -tunlp` depending on what tools you have installed and it will list the "listening" ports and 47808 should not be there.

closed 2023-09-19 by mohakbhalla7