JoelBender/bacpypes #478: Help on setting up alarm on virtual device

khalaniaiyaj, 2023-01-18:
Is there any way or example using which we can setup alarm on specific object for virtual device and subscribe to it in YABE?

JoelBender, 2023-01-19:
I'm not sure about the YABE subscription part, but I'll keep this issue open for an additional sample application.  Do you have a specific object type in mind?

khalaniaiyaj, 2023-02-02:
Hi @JoelBender 

so basically i have create following sample object 

```test_ai = SampleAnalogValueObject(
        objectIdentifier=("analogInput", 1),
        objectName="ai",
        presentValue=0.0,
        covIncrement=0.5,
        eventDetectionEnable=True,
        eventEnable=[1, 1, 1],
        ackedTransitions=[1, 1, 1],
        notifyType=0,
        reliability=1,
        outOfService=False,
        eventState="normal",
        statusFlags=StatusFlags(),
        units=19,
    )
    
 
I am able to subscribe to the object from Yabe explore and able to monitor the changes and based on the new value, i am able to change the eventState and statusFlags. But wanted to know if we change the status to ALARM, how can we implement AlarmSummaryRequest in our device so that we can share the summary of the alarm to client?