JoelBender/bacpypes #479: ini settings from database rather than ini file

kheldaroz402, 2023-01-27:
HI Joel,

is it possible to set the ini settings through the program rather than ini file. I have the settings in a database and I would rather use those than having to edit a ini file when I need them to change

JoelBender, 2023-01-27:
Imagine this piece of code:
```python
    # make a device object
    this_device = LocalDeviceObject(
        objectName=args.ini.objectname,
        objectIdentifier=int(args.ini.objectidentifier),
        maxApduLengthAccepted=int(args.ini.maxapdulengthaccepted),
        segmentationSupported=args.ini.segmentationsupported,
        vendorIdentifier=int(args.ini.vendoridentifier),
        )
```
Translated into this:
```python
    config = {
        "objectName": "Algol",
        "objectIdentifier": 999,
        "maxApduLengthAccepted": 1024,
        "segmentationSupported": "segmentedBoth",
        "vendorIdentifier": 555,
    }

    # make a device object
    this_device = LocalDeviceObject(**config)
```
And then get your config object from the database, perhaps a JSON text blob.  You'll still have to restart the application when the configuration changes, there is no easy way to abstract the objects added to an application to live in the database.

kheldaroz402, 2023-01-30:
Thanks Joel :)

kheldaroz402, 2023-01-31:
My code for others to use / improve :)

`mydbProgram = MySQLdb.connect(host=os.getenv('dbhost'),
                              user=os.getenv('dbuser'),
                              passwd=os.getenv('dbpass'),
                              database=os.getenv('WebAIProgramDB'))

mycursorProgram = mydbProgram.cursor(MySQLdb.cursors.DictCursor)

mycursorSettings.execute("SELECT * FROM config")

myresult = mycursorSettings.fetchall()
WebAIConfig = {}
for row in myresult:
    #logging.debug(row)

    WebAIConfig[row['parameter']] = row['value']

""" Get BACnet Settings """
objectName = WebAIConfig['SystemName']
objectIdentifier = WebAIConfig['BACnetID']
maxApduLengthAccepted = WebAIConfig['maxApduLengthAccepted']
segmentationSupported = WebAIConfig['segmentationSupported']
vendorIdentifier = WebAIConfig['BACnet_vendorID']
vendorName = WebAIConfig['BACnet_vendorName']
modelName = WebAIConfig['ControllerType']
firmwareRevision = WebAIConfig['version']
serialNumber = WebAIConfig['QRcodeWebAI']
location = WebAIConfig['Location']
foreignPort = WebAIConfig['foreignPort']
foreignBBMD = WebAIConfig['foreignBBMD']
foreignTTL = WebAIConfig['foreignTTL']
max_masters = WebAIConfig['BACnet_MAX_Master']
maxinfo = WebAIConfig['BACnet_maxinfo']

config = {
    "objectName": objectName,
    "objectIdentifier": int(objectIdentifier),
    "maxApduLengthAccepted": int(maxApduLengthAccepted),
    "segmentationSupported": segmentationSupported,
    "vendorIdentifier": vendorIdentifier,
    "vendorName": vendorName,
    "modelName": modelName,
    "firmwareRevision": firmwareRevision,
    "serialNumber": serialNumber,
    "location": location,
    "foreignPort": foreignPort,
    "foreignBBMD": foreignBBMD,
    "foreignTTL": foreignTTL,
}

mycursorProgram.execute("SELECT * FROM Com_Port_Status where Display = 'BACnet'")

myresult = mycursorProgram.fetchall()
# gets the number of rows affected by the command executed
row_count = mycursorProgram.rowcount
print("number of affected rows: {}".format(row_count))
if row_count == 0:
    print(color.PURPLE + "mstp not Enabled" + color.END)
    mstpEnable = False
else:
    print(color.PURPLE + "mstp Enabled" + color.END)
    mstpEnable = True
    for row in myresult:
        address = row['IP_Address']
        interface = '/dev/' + row['Port']
        baudrate = row['baudrate']

# application interval is refresh time in minutes (default 5)APPINTERVAL = int(os.getenv("APPINTERVAL", 5)) * 60 * 1000
APPINTERVAL = int(os.getenv("APPINTERVAL", 1)) * 0.1 * 1000
APPINTERVAL2 = int(os.getenv("APPINTERVAL", 5)) * 60 * 1000

# dictionary of names to objects
objects = {}
# globals
property_list = None

# make a device object
mstp_args = {
 '_address': int(address),
 '_interface':str(interface),
 '_max_masters': int(max_masters),
 '_baudrate': int(baudrate),
 '_maxinfo': int(maxinfo),
}

# make a device object
this_device = LocalDeviceObject(**config, **mstp_args)`

closed 2023-01-31 by kheldaroz402