JoelBender/bacpypes #482: Unable to get response in subscribeCOV

ajhwb, 2023-03-09:
Hi Devs,

I was testing a BACnet device Sierra Monitor SMC FS-QS-3010
It's in the same network of my workstation

The BACnet device is: 192.168.2.105
Workstation: 192.168.2.100

I was able to query a whois and get a broadcast message with sample/WhoIsIAm.py

```
(venv) python samples\WhoIsIAm.py
> whois
> pduSource = <RemoteStation 5:0x00000005ef89>
iAmDeviceIdentifier = ('device', 389001)
maxAPDULengthAccepted = 1458
segmentationSupported = noSegmentation
vendorID = 37
pduSource = <RemoteStation 5:0x00000005ef8a>
iAmDeviceIdentifier = ('device', 389002)
maxAPDULengthAccepted = 1458
segmentationSupported = noSegmentation
vendorID = 37
pduSource = <RemoteStation 5:0x00000005ef8b>
iAmDeviceIdentifier = ('device', 389003)
maxAPDULengthAccepted = 1458
segmentationSupported = noSegmentation
vendorID = 37
pduSource = <RemoteStation 5:0x00000005ef8c>
iAmDeviceIdentifier = ('device', 389004)
maxAPDULengthAccepted = 1458
segmentationSupported = noSegmentation
vendorID = 37
```

However, when testing a COV subscription, i didn't get any response, the query was sent (captured in Wireshark) 
[https://imgur.com/4NbdtXh](https://imgur.com/4NbdtXh)

The sample/COVClient.py debug output is:

```
venv) python samples\COVClient_Mod.py --debug
DEBUG:__main__:initialization
DEBUG:__main__:    - args: Namespace(buggers=False, debug=[], color=None, route_aware=None, ini={'objectname': 'Betelgeuse', 'address': '192.168.2.100/24:47808', 'objectidentifier': '599', 'maxapdulengthaccepted': '1024', 'segmentationsupported': 'segmentedBoth', 'vendoridentifier': '15', 'foreignbbmd': '192.168.2.1', 'foreignttl': '30'})
DEBUG:__main__:    - this_device: <bacpypes.local.device.LocalDeviceObject object at 0x000001F806D5C350>
DEBUG:__main__.SubscribeCOVApplication:__init__ (<bacpypes.local.device.LocalDeviceObject object at 0x000001F806D5C350>, '192.168.2.100/24:47808')
DEBUG:__main__:    - this_console: <SubscribeCOVConsoleCmd(ConsoleCmd, started 7916)>
DEBUG:__main__:running
> subscribe 192.168.2.105/24:47808 96 binaryOutput:1 true 120
DEBUG:__main__.SubscribeCOVConsoleCmd:do_subscribe ['192.168.2.105/24:47808', '96', 'binaryOutput:1', 'true', '120']
DEBUG:__main__.SubscribeCOVConsoleCmd:    - issue_confirmed: True
DEBUG:__main__.SubscribeCOVConsoleCmd:    - lifetime: 120
DEBUG:__main__.SubscribeCOVConsoleCmd:    - request: <bacpypes.apdu.SubscribeCOVRequest(5) instance at 0x1f8047f6190>
    <bacpypes.apdu.SubscribeCOVRequest(5) instance at 0x1f8047f6190>
        pduDestination = <Address 192.168.2.105>
        pduExpectingReply = 1
        pduNetworkPriority = 0
        apduType = 0
        apduService = 5
        subscriberProcessIdentifier = 96
        monitoredObjectIdentifier = ('binaryOutput', 1)
        issueConfirmedNotifications = True
        lifetime = 120
        pduData = x''
DEBUG:__main__.SubscribeCOVConsoleCmd:    - iocb: <bacpypes.iocb.IOCB(1) instance at 0x1f8047f64d0>
    <bacpypes.iocb.IOCB(1) instance at 0x1f8047f64d0>
        args = (<bacpypes.apdu.SubscribeCOVRequest(5) instance at 0x1f8047f6190>,)
        kwargs = {}
        ioState = 0
        ioComplete = <threading.Event at 0x1f806d869d0: unset>
        ioCallback = []
        ioPriority = 0
DEBUG:__main__.SubscribeCOVConsoleCmd:    - error: <bacpypes.apdu.AbortPDU(None,1) instance at 0x1f806d87a90>
    <bacpypes.apdu.AbortPDU(None,1) instance at 0x1f806d87a90>
        pduSource = <Address 192.168.2.105>
        pduExpectingReply = 0
        pduNetworkPriority = 0
        apduType = 7
        apduSrv = False
        apduInvokeID = 1
        apduAbortRejectReason = 65
        pduData = x''
```

But when testing with Yabe (Yet another bacnet explorer), same workstation and network, the COV subscription seem to have responses in Wireshark capture as follow:

[https://imgur.com/0v3iMBJ](https://imgur.com/0v3iMBJ)
[https://imgur.com/kxc3rHk](https://imgur.com/kxc3rHk)

Here my BACPypes.ini

```
[BACpypes]
objectName: Betelgeuse
address: 192.168.2.100/24:47808
objectIdentifier: 599
maxApduLengthAccepted: 1024
segmentationSupported: segmentedBoth
vendorIdentifier: 15
```

I presume, I have issue with sending/receiving unicast response message from the BACnet device or could be configuration etc.
Can I have any pointers on this issue?

Thanks!

JoelBender, 2023-03-09:
Your destination address is incorrect, you are sending the request to the router rather than through the router to the device.  The capture dumps aren't showing you the NPDU source or destination fields.

> subscribe 5:0x00000005ef8b 96 binaryOutput:1 true 120

ajhwb, 2023-03-10:
Thank you Joel!
I'm now able to subscribe using the correct address.

```
> subscribe 5:0x00000005ef8b 91 analogOutput:192 true 120
> ('analogOutput', 192) changed

    presentValue is 14.0
    statusFlags is [0, 0, 0, 0]
('analogOutput', 192) changed

    presentValue is 15.0
    statusFlags is [0, 0, 0, 0]    
```

closed 2023-03-10 by ajhwb