JoelBender/bacpypes #483: WhoIsIAm.py only able to get response from the BACnet Simulator in the first 1-2 minutes.

11208, 2023-03-29:
I am trying out the WhoIsIAm.py and only able to get IAm response from the BACnet Simulator on the first 1-2 minutes.
I am using YABE and SoftDEL BOSS as a simulators.

Steps:
1. Start YABE and SoftDEL BOSS.
2. Run WhoIsIAm.py and key in whois command and got response
[image]
[image]
 
3. Wait for awhile and key in whois again and no response
[image]
[image]

4. Restart YABE simulator and key in the whois command again got response
[image]
[image]

I dont know why after awhile WhoIsIAm.py cant get the IAM response.

JoelBender, 2023-03-30:
Can you please try and reproduce this with the `WhoIsIAm.py` sample application?

ericprom, 2023-03-30:
> Can you please try and reproduce this with the `WhoIsIAm.py` sample application?

That is a WhoIsIAm.py. I just renamed the file. How can I fix this, when I can not restart the bacnet divice situation? Is the issue with the Simulator or Bacpypes? Form the Wireahark, Bacpypes did send Who-Is, but no device return I-Am.