JoelBender/bacpypes #484: How can I provide an address of a device with IP,Port,network, and mac address

venusjain2003, 2023-04-19:
objective: I need to read a property of a device, which is behind a router.
What I need: I need to know how to provide the complete address (including IP, Port, Number and Mac Address) to the "read" command of the ReadWriteProprty.py console application.

Approach that I am taking:
1. Start a local application ReadWriteProprty.py with a localdeviceobject on my local machine.
2. Fire a read command to the device behind a router on my orgnization's network. I am connected to that network using VPN.

Thanks
Venus

JoelBender, 2023-04-20:
There are some challenging parts to this problem, taking the last bit first.

Because you are using a VPN connection the BACnet address used by your application needs to be the one provided to you by the VPN server and provided to your workstation as an interface and address.  For example, my device in my office has an IPv4 address on **eth0** has something like 192.168.0.15/24 but when authenticated with the VPN server there is an additional interface **csctun0** (Cisco AnyConnect Client) with an address like 10.0.99.16/24.

You are probably not going to be connected to a network that has a BBMD, so for broadcast traffic (like Who-Is-Router-To-Network) your application needs to register as a foreign device with an existing BBMD and you need to use the foreign device version of the application.

Assume you need to send a read request to MAC address 12 on BACnet network 75 and there is an IPv4 router to that network at address 10.0.75.17.  The usual address form is `75:12` and the library will find the router, bind the router address to the network, and then send a unicast message to the router saying please forward this packet.  If you register as a foreign device the library will send the routing request wrapped inside a Distribute-Broadcast-To-Network message to the BBMD which will forward it.

If for some reason you don't want to or can't register as a foreign device, there is a "route aware" option you can turn on and then specify the router directly in the address, in this case `75:12@10.0.75.17`.  The routing table lookup is bypassed and your request is unicast directly to the router.

venusjain2003, 2023-04-20:
Awesome! thank you so much for this information @JoelBender . I'll try the "route aware" option and let you know.