JoelBender/bacpypes #486: Invalid constructor datatype when writing a recipient list to a notification class object

mosterme1003, 2023-05-11:
Hey! I have some trouble writing a `recipientList` to a `notificationClass` on another BACnet device in my subnet. BACnet wants a BACnetList of BACnetDestination so I tried to implement it with `ListOf(Destination)` like you did in your `EventNotification.py` sample.

```python
class NotificationClass:

    ListOfDestination = ListOf(Destination)

    def build_write_request(self, ncIp, ncId, listRecipients):
        request = WritePropertyRequest(
            objectIdentifier=('notificationClass', ncId),
            propertyIdentifier='recipientList',
            propertyValue=Any(),
        )
        request.pduDestination = Address(ncIp)
        request.propertyValue.cast_in(listRecipients)

        return request

    def write_value(self, bacnet, request, timeout=60):
        try:
            iocb = IOCB(request)
            iocb.set_timeout(timeout)
            deferred(bacnet.this_application.request_io, iocb)

            iocb.wait()

            if iocb.ioResponse:
                apdu = iocb.ioResponse

                if not isinstance(apdu, SimpleAckPDU):
                    print("Not an ack, see debug for more infos.")
                    return

            if iocb.ioError:
                apdu = iocb.ioError
                reason = find_reason(apdu)
                raise NoResponseFromController(
                    "APDU Abort Reason : {}".format(reason))

        except WritePropertyException as error:
            print(("exception: {!r}".format(error)))

    def encode_listRecipients(self, listDeviceIds):

        listNewRecipients = []
        for deviceId in listDeviceIds:
            destination = Destination(
                validDays=[1, 1, 1, 1, 1, 1, 1],
                fromTime=[0, 0, 0, 0],
                toTime=[23, 59, 59, 99],
                recipient=Recipient(device=('device', int(deviceId))),
                processIdentifier=0,
                issueConfirmedNotifications=True,
                transitions=[1, 1, 1],
            )
            listNewRecipients.append(destination)

        listRecipients = self.ListOfDestination(listNewRecipients)

        return listRecipients

    def write_notificationRecipients(self, bacnet, ncIp, ncId, listDeviceIds):
        try:
            listRecipients = self.encode_listRecipients(listDeviceIds)
            request = self.build_write_request(ncIp, ncId, listRecipients)
            self.write_value(bacnet, request)
        except Exception as error:
            print(traceback.format_exc())
            return {}

if __name__ == "__main__":
    bacnet = BAC0.connect(ip="192.168.19.176")
    nc = NotificationClass()
    nc.write_notificationRecipients(bacnet, "10.64.134.164", 1, ["90008", "90009"])
```

Now I get a TypeError while casting the destination list into the request's property value:
```
Traceback (most recent call last):
  File "~/Documents/Python/bacnet-tool-em/functions/notificationClass.py", line 86, in write_notificationRecipients
    request = self.build_write_request(ncIp, ncId, listRecipients)
  File "~/Documents/Python/bacnet-tool-em/functions/notificationClass.py", line 25, in build_write_request
    request.propertyValue.cast_in(listRecipients)
  File "C:\Program Files\Python379\lib\site-packages\bacpypes\constructeddata.py", line 1291, in cast_in
    element.encode(t)
  File "C:\Program Files\Python379\lib\site-packages\bacpypes\constructeddata.py", line 607, in encode
    value.encode(taglist)
  File "C:\Program Files\Python379\lib\site-packages\bacpypes\constructeddata.py", line 111, in encode
    helper = element.klass(value)
  File "C:\Program Files\Python379\lib\site-packages\bacpypes\primitivedata.py", line 1568, in __init__
    raise TypeError("invalid constructor datatype")
TypeError: invalid constructor datatype

exception: TypeError('invalid constructor datatype')
```

Did I do something wrong? I thought that `ListOf(Destination)` would be the correct datatype but maybe I'm wrong or it's just a small mistake. I would really appreciate your help. Thank you very much!

JoelBender, 2023-05-11:
The `Time` primitive takes a four element tuple, not a four element list, as an `__init__()` parameter.
```python
    ...
    fromTime=(0, 0, 0, 0),
    toTime=(23, 59, 59, 99),
    ...
```

mosterme1003, 2023-05-12:
It works fine now. Thank you!

closed 2023-05-12 by mosterme1003