JoelBender/bacpypes #489: Strange error in apdu.encode(xpdu)

mayeranalytics, 2023-05-26:
Hi Joel,

I'm getting a very strange error in apdu.encode(xpdu):

```text
Traceback (most recent call last):
  File "/home/pi/miniconda3/lib/python3.4/site-packages/bacpypes/appservice.py", line 1486, in sap_indication
    apdu.encode(xpdu)
  File "/home/pi/miniconda3/lib/python3.4/site-packages/bacpypes/apdu.py", line 695, in encode
    Sequence.encode(self, self._tag_list)
  File "/home/pi/miniconda3/lib/python3.4/site-packages/bacpypes/constructeddata.py", line 115, in encode
    helper.encode(tag)
  File "/home/pi/miniconda3/lib/python3.4/site-packages/bacpypes/primitivedata.py", line 1810, in encode
    tag.set_app_data(Tag.objectIdentifierAppTag, struct.pack('>L', self.get_long()))
struct.error: argument out of range
```

`get_long` returns a number that is too large to pack into ">L" (32 bits).
I have no clue what's going on. Do you have any ideas?

Many thanks

JoelBender, 2023-05-26:
Inside `get_long()` is this:

```python
        objType, objInstance = self.get_tuple()

        # pack the components together
        return ((objType << 22) + objInstance)
```

So this error is (a) the object instance is greater than 4294967295, or (b) the object type is greater than 1023.

mayeranalytics, 2023-05-27:
Yeah.... the dear user switched object instance and object type 🤨 - safety measures added.
Thanks for the pointer!