JoelBender/bacpypes #490: Using MultipleReadProperty and YABE Room Simulator

ind-georghuber, 2023-05-26:
Hi,

i want to test the MultipleReadProperty sample with YABE room simulator. 
`point_list = [
    ('10.0.0.34', 'analogValue:1', 'presentValue'),
    ]`

When i start the script i get after some time the answer "noResponse".
In the attached trace you can see that there is a response coming. It seems like that the IOCB dont get this ACK.
Tested it on a Windows VM with python 3.11.3.

Can you help me with this?
Thank you

[image]

JoelBender, 2023-05-26:
Please follow this up with a picture of the Complex-ACK that was returned, like number 457.  There may be something more in the BVLL or NPDU layers that is preventing the request and response from lining up.  The `[1]` being repeated is due to the number of APDU retries and the delay between request attempts is due to the APDU timeout.

ind-georghuber, 2023-05-27:
Yes of course, here it is. Hope it helps.

[image]

JoelBender, 2023-05-28:
In the bacpypes-pcap project there is a **hexdecode.py**.  Here is the request:
```bash
$ python hexdecode.py 00505601b9e20050.5691796008004500.002d64ed00008011.00000a0000210a00.0022bac0bac00019.146d810a00110104.0244010c0c008000.011955
    pduSource = <Address 10.0.0.33>
    pduDestination = <Address 10.0.0.34>
    pduExpectingReply = True
    pduNetworkPriority = 0
    apduType = 0
    apduSeg = False
    apduMor = False
    apduSA = True
    apduMaxSegs = 4
    apduMaxResp = 4
    apduService = 12
    apduInvokeID = 1
    objectIdentifier = ('analogValue', 1)
    propertyIdentifier = 'presentValue'
    pduData = x''
```
And here is the response:
```bash
$ python hexdecode.py 00505691796000505691b9e208004500.003377fc00008011.ae7b010000220a00.0021c9f6bac0001f.cfc4810a00170100.30010c0c00800001.19553e4441a80000.3f
    pduSource = <Address 0x01000022c9f6>
    pduDestination = <Address 10.0.0.33>
    pduExpectingReply = False
    pduNetworkPriority = 0
    apduType = 3
    apduSeg = False
    apduMor = False
    apduService = 12
    apduInvokeID = 1
    objectIdentifier = ('analogValue', 1)
    propertyIdentifier = 'presentValue'
    propertyValue
            <bacpypes.primitivedata.Tag(real) instance at 0x7f43a06bd7b0>
                tagClass = 0 application
                tagNumber = 4 real
                tagLVT = 4
                tagData = '41.a8.00.00'
    pduData = x''
```
The clue for me that they don't line up is that the `ba c0 ba c0` isn't in the UDP header of the response.  I think that if you expand the IP and UDP headers you should see they aren't a matched pair.

ind-georghuber, 2023-05-30:
Yeah thats true. Thank you, udp destination was the standard port 47808. But the Server listens on 51702. When i change the Address to 10.0.0.34:51702, it works and i get the value. ;D

closed 2023-05-30 by ind-georghuber