JoelBender/bacpypes #491: Question on How to Make a Virtual BACnet Device that Doesn't Understand "Real" Values

wehuang16, 2023-06-16:
I have a question. I am making a virtual BACnet device using the “mini_device.py” file here (https://github.com/JoelBender/bacpypes/blob/master/samples/mini_device.py). The “mini_device.py” file defines BACnet objects such as “test_av” and “test_bv”. Each BACnet object has several properties; for example,  “test_av” has properties “objectIdentifier”, “objectName”, “presentValue”, “statusFlags”, and “covIncrement”. Let’s say one of these properties is “Writable” (defined by the class “WritableProperty”) (see “class WritableProperty(StandardProperty)” in the “object.py” file at https://github.com/JoelBender/bacpypes/blob/master/py34/bacpypes/object.py). Let’s say the property that is “Writable” has a value which is originally defined as an “Unsigned8” (see “class Unsigned8(Unsigned)” at https://github.com/JoelBender/bacpypes/blob/master/py34/bacpypes/primitivedata.py). When I use an external Energy Management Information Systems (EMIS) tool to write a “Real” (as opposed to an Unsigned8) value to the “Writable” property in the  virtual BACnet device I made using the “mini_device.py” file, I see that the “Writable” property has the “Real” value that I write back to (no longer an Unsigned8 value). 

My question is, is it possible for me to make a virtual BACnet device by writing the “mini_device.py” file in a way that the “Writable” property in the virtual BACnet device doesn’t understand “Real” values at all and only understand “Unsigned8” values (that is, when I write a  “Real” value, such as 3.0000000, to the “Writable” property in the virtual BACnet device, the “Writable” property will take the “Real” value, but converts the “Real” value to an “Unsigned8” value in an unexpected way, such as converting a “Real” value 3.0000000 to an “Unsigned8” value of 2. Then, when I use the EMIS tool to read the value from the “Writable” property, I will get an “Unsigned8” value of 2.)? If so, how would the coding look like in “mini_device.py”?

JoelBender, 2023-06-17:
BACnet is has strict typing, for an Analog Value Object (Clause 12.4) the present value is going to be REAL.  By substituting `IntegerValueObject` for `AnalogValueObject`, giving it an an object identifier with the appropriate type `('integerValue', 1)` you will be able to read and write integer values.  Notice that there is also a `PositiveIntegerValueObject` class which is closer to the Unsigned8 datatype.

That's half the problem, the other half is to verify that the EMIS tool understands these types.

wehuang16, 2023-06-21:
Hey Joel, thank you so much for replying to me! That’s very valuable information!

I have an additional question shown below.

I have modified the original “mini_device.py” document  (https://github.com/JoelBender/bacpypes/blob/master/samples/mini_device.py) to create a new “mini_device.py” document (https://drive.google.com/file/d/1UMqEpSYbTJCfULEb9GIH8_y6zPfdjqvl/view?usp=drive_link ). I added the “PositiveIntegerValueObject” and “integerValueObject” to the  new “mini_device.py” document; then, I used the new “mini_device.py” document to create a “virtual BACnet device” on my laptop. Next, I used the EMIS to try to read the “PositiveIntegerValueObject” and “integerValueObject”, but the EMIS tool didn’t seem to recognize either of these 2 objects. Then, I added the “MultiStateValueObject” to the new “mini_device.py” document (also see screenshot below):
[image]

I was able to use the EMIS tool to read the “MultiStateValueObject”! I read from the EMIS tool that the value “numberOfStates” was an “unsigned” value of 1. I used a software called Wireshark to examine the BACnet network traffic, and Wireshark showed that the value of 1 was indeed an “unsigned” value.

To examine that EMIS is writing to the “numberOfStates” value, I actually needed to change the source code of Bacpypes. The change is shown below for the document “object.py” (https://github.com/JoelBender/bacpypes/blob/master/py34/bacpypes/object.py), where I changed the “OptionalProperty” to “WritableProperty” for the variable “numberOfStates” .

[image]
[image]

Then, I used the EMIS tool to write a different value (for example, 2) to the “numberOfStates” variable, and used the EMIS tool to read back the value of the “numberOfStates” variable. I used the Wireshark software to verify that the number 2 that I used the EMIS to write to the virtual BACnet device and read back from the BACnet device is an “unsigned” value. 

This is a success! I then used the EMIS tool to write a value of 3.4 to the “numberOfStates” variable and read back the value of the “numberOfStates” variable from the virtual BACnet device. I found that the value of the “numberOfStates” variable reading back from the virtual BACnet device becomes an “unsigned” value of 3 (verified using Wireshark software)! The summary was that the “MultiStateValueObject” was able to RESTRICT that the value needed to be “unsigned”.

My goal now is to use the EMIS tool to read and write a “ScheduleObject”. The “ScheduleObject” is defined in the new “mini_device.py” document as follows:
[image]

I first went to the source code document “object.py” (https://github.com/JoelBender/bacpypes/blob/master/py34/bacpypes/object.py), where I changed the “ReadableProperty” to “WritableProperty” for the variable “weeklySchedule”. Then, I used the EMIS tool to read the variable “weeklySchedule” from the virtual BACnet device, and I used the Wireshark software to see that the value of “TimeValue” is an “unsigned” integer. However, when I used the EMIS tool to write a completely new “weeklySchedule” to  the virtual BACnet device and read back the new “weeklySchedule” from the virtual BACnet device, the value of “TimeValue” became a “real” number. In summary, the “scheduleObject” was NOT able to RESTRICT that the value from the “TimeValue” of the “weeklySchedule” must be “unsigned”.

I then traced back to the properties of the “DailySchedule”, and changed the “AnyAtomic” to “Unsigned”:
[image]

[image]
[image]
However, this caused a “TypeError: invalid constructor datatype”. 

My question is, is there a way to modify the Bacpypes code somehow, so that I could make the “scheduleObject” to restrict the value of “TimeValue” to only use “unsigned” values but not “Real” values, just like the “MultiStateValueObject” could?

I would like to share some background information of why I wanted to restrict the “scheduleObject” to only use “unsigned” values. I wanted to use the EMIS tool to write new schedules to a physical Building Automation System (BAS). When using the EMIS tool to write to a “MultiStateValueObject” , the EMIS tool was able to write “unsigned” values to the “MultiStateValueObject”. However, due to the limitations of the EMIS tool that I was not able to change, the EMIS tool was only able to write “Real” values to a “scheduleObject”. However, when it comes to the “scheduleObject”,  the physical BAS system happened to not understand “Real” values at all, and the BAS only understood multi-state values (a type of “unsigned” value). Thus, when writing a “Real” value of 3.00000 to the BAS from the EMIS, the BAS will interpret the  “Real” value 3.00000 to be an “unsigned” value of 2 (the intention was to get the BAS to have an “unsigned” value of 3). Since I only have access to the EMIS and doesn’t have access to the BAS, I would like to use Bacpypes’ “mini_device.py” to emulate the exact same behavior of the physical BAS (only understand “unsigned” values but don’t understand “Real” values), so that I could do further testing with the EMIS tool that I can access.

JoelBender, 2023-06-22:
You can create your own MultiStateValueObject by creating a subclass and overriding properties rather than change the library:
```python
class MyMultiStateValueObject(MultiStateValueObject):
    properties = \
        [ WritableProperty('presentValue', Unsigned)
        ]
```
There might be a problem with your use case if you need to be able to read and write zero (0) because the _present-value_ property must be greater than zero.  For your **Unsigned8** you need 255 states.
```python
    # make a multistate value object
    msvo = MyMultiStateValueObject(
        objectIdentifier=('multiStateValue', 1),
        objectName='My Special Object',
        presentValue=1,
        numberOfStates=255,
        stateText=[("%d" % i) for i in range(1,256)],
        )
    _log.debug("    - msvo: %r", msvo)
```
Using the _present-value_ is more in line with the intent of the object rather than _number-of-states_, which is also restricted to be greater than zero.

> My question is, is there a way to modify the BACpypes code somehow, so that I could make the “scheduleObject” to restrict the value of “TimeValue” to only use “unsigned” values but not “Real” values, just like the “MultiStateValueObject” could?

There is an added complication with filtering `AnyAtomic` because the values are stuffed into a wrapper and because `TimeValue` is inside a `DailySchedule` and a `SpecialEvent` which are both inside arrays.  What kind of service is the EMIS using to write to a schedule object?  When the EMIS uses Real values to update the weekly schedule does the BAS return an error or just blindly convert the Real to an Unsigned and send back a simple ack?

Because of the way `AnyAtomic` is referenced in the rest of the constructed data code, it's easier in this case to patch the `AnyAtomic` code to add/replace the type checking.  I attached an **untested** version of constructeddata.py.
[constructeddata.txt](https://github.com/JoelBender/bacpypes/files/11828576/constructeddata.txt)

wehuang16, 2023-06-22:
Hey Joel!  Regarding the service that the EMIS uses, the EMIS uses a BACnet function/protocol called "bacnetWriteSchedule()" to write schedules back to the BAS. I would input the object name ("SCH1", which means a schedule object), object property (which will be "weekly_schedule"), the new schedule I want to write to BAS, and the priority level to the "bacnetWriteSchedule()" function. This is all I know about the EMIS service.

When the EMIS uses Real values to update the weekly schedule to the BAS, the BAS just blindly convert the Real to an Unsigned and send back to EMIS. The BAS blindly convert the Real to an Unsigned in a strange way (would convert a Real value of 3.000 to an Unsigned value of 2, rather than convert a Real value of 3.000 to an Unsigned value of 3).

Thank you so much for taking the extra effort to create a patch code for AnyAtomic and sharing how I could define my own multiStateValueObject!

JoelBender, 2023-06-23:
> ...EMIS uses a BACnet function/protocol called "bacnetWriteSchedule()" to write schedules back to the BAS.

Ah, more black boxes!  Rather than throwing an error you could update that `AnyAtomic` code to replace the Real value with some Unsigned using whatever algorithm you'd like.

> The BAS blindly convert the Real to an Unsigned in a strange way...

I've seen variations of this on platforms that don't support floating point numbers, but that was in the [8080 days](https://retrotechnology.com/herbs_stuff/float.html) before the 8087 and that was a _long_ time ago!

wehuang16, 2023-06-23:
Thanks again for your information and help!

closed 2023-07-21 by wehuang16