JoelBender/bacpypes #493: Question regarding Max APDU

christopherjohnson1535, 2023-06-26:
Hello, I have a few questions regarding max APDU and BACpypes.

1. Does BACpypes take care of ensuring that the APDU is within the limit of the BACnet device? I.e. Can I request infinite properties using readMultipleProperties? Or is it my responsibility to ensure that I am not requesting too many properties at once?
2. Does using the local device object affect the amount of properties that will be sent in a request? For example, if I use a local device with a max APDU of 480, will it only send our messages with an APDU of 480?

Thanks!

JoelBender, 2023-06-28:
1. No, it assumes an infinite amount of buffer space for encoding and decoding packets.  For example, if this is larger than what can be contained in a single UDP packet (limited by the next layer down, like Ethernet of 1.5K (or 9K with jumbo frames)) then the packet is fragmented, which might not be supported by the peer.  There could be a platform that limits the packet size in the socket library implementation so you'll get an exception of some kind. The limit of the number of properties that you can successfully request will most likely be determined by the server (a.k.a. B-side) with its maximum APDU size and segmentation support.
2. Also no.

christopherjohnson1535, 2023-06-29:
Do you have any recommendations of a clean way to ensure that not too many properties are requested beside testing with different property counts?