JoelBender/bacpypes #502: [Question] Proper BBMD.py usage

ChrisM09, 2023-09-24:
Hello,

I will preface this with saying that I'm still basically new to BACnet and especially this library.

I will also say that I tried to add some debugging capabilities, but was wondering if adding the `--debug <bacpypes.<service>` flag was the only way besides the infamous print statement? But back to the main issue.

# Main Issue

**Summary:** I am trying to see the behavior of the BBMD programs with respect to writing and reading, but I'm having trouble seeing the actual `writebdt` command being carried out and am wondering if this is user error, or if I need to do further reading on the docs for the program. 

## Setup
I'll give the scenario.

1. Machine 1 (M1) - IP X.X.X.X     // This would be the BBMD machine
2. Machine 2 (M2) - IP Y.Y.Y.Y     // This would be a machine reading/writing to/from the BBMD

## Steps carried out + behavior observed
When I run the *BBMD.py* program with `python3 samples/bbmd.py X.X.X.X 1.2.3.4/24`, it seems to start up just fine with `1.2.3.4/24` inside its BDT.

Next, I run the *ReadWriteBBMD.py* program on *M2* with `python3 samples/ReadWriteBBMD.py Y.Y.Y.Y` in which I'm presented with the sub-console (for the lack of a better term). When I run `readbdt X.X.X.X`, it returns the expected table with the entry 1.2.3.4 present within.

Now, the problem comes when I use `writebdt`. Maybe I'm using it wrong. But if I run the command (within the sub-console) `writebdt X.X.X.X 2.3.4.5/24`, I can confirm that it didn't work by running a `readbdt X.X.X.X`  command on *M2*  and the response does not have the additional entry added into M1's BDT.

Hopefully it's a simple case of user error (and hopefully a relatively simple fix). But if not, I would love some guidance on this. I was looking through previous issues and tried a few of the older solutions, but to no avail. I guess a simple follow up question is "does the BBMD" need a legitimate IP that it can reach? 

Any help is appreciated.
Thank you so much.

JoelBender, 2023-09-24:
> if adding the --debug <bacpypes.<service> flag was the only way besides the infamous print statement?

You can use any tool to debug Python applications you would like, including [pdb](https://docs.python.org/3/library/pdb.html) or one [built into an IDE](https://code.visualstudio.com/docs/python/debugging).  I added the `--debug` feature to work on long running applications that ran as a daemon, the `print >> log, "test"` syntax is in the dustbin of history.  The `if _debug:` thing is because making the logging handler calls was taking _waaay_ too long when there was no handler attached, 500MHz Pentium 3 days.

> Maybe I'm using it wrong.

No, it's not implemented.  When you sent the request it responded with a non-zero result code (0x10, decimal 16) and [in the code](https://github.com/JoelBender/bacpypes/blob/master/py34/bacpypes/bvllservice.py#L814-L822) for the `BIPBBMD.confirmation()` method it builds and returns the `Result(code=0x0010, user_data=pdu.pduUserData)` PDU.

> I guess a simple follow up question is "does the BBMD" need a legitimate IP that it can reach?

No, it's just a table, and you (or other devices!) can put in anything you'd like.  This was actually a serious problem because some devices from a vendor thought it was doing me a favor by jamming updated BDT's in all the other ones it could discover leading to massive broadcast floods.

Patches/merge requests always welcomed.  Maybe there should be an option that you can turn on to make it writable, but it off by default?  The Network Port Object demands it to be writable:

> **Clause 12.56.34 BBMD_Broadcast_Distribution_Table**
> This property, of type BACnetLIST of BACnetBDTEntry, is required to be present and writable if BACnet_IP_Mode is
> BBMD.

ChrisM09, 2023-09-25:
Ahhh.. I see. Thank you so much for the information!

I've read some of the standard and am slightly confused as to where the handling of this request should be.

I'm assuming that the code that will handle the request should be in the `BIPBBMD.confirmation` method since it handles the type checking for the message (specifically the writebdt class)? My understanding is that an `indication` is used for the _invocation_ of a method to handle the request whereas a `confirmation` is used to represent the _completion_ of a method; so this is where my confusion comes from. Maybe I need to read the sentence a few more times in a different, way but figured I'd ask.

Thank you!

JoelBender, 2023-09-26:
The way I put the stack together, requests go down and turn into indications, responses go up and turned into confirmations.  The BIPBBMD is a "client" of the UDP layer as a "server" and all upstream messages appear as `confirmation()` calls, even if they are requests.

For the network layer and application layers I created something like a motorcycle sidecar, the motorcycle being the service access point and the sidecar being the service element (the classes are in the `bacpypes.comm` module).  The intent is so that network layer applications can make requests of other network layer applications, like two sidecars (one on the left and one on the right) passing notes back and forth while their motorcycles are cruising down the road.  The `NetworkServiceAccessPoint` has to decide if upstream packets from the link layer (or virtual link layer in the case of B/IP) are _network application layer_ requests, passing it to the `NetworkServiceElement`, or should continue upstream to the application layer.

I did not follow this model for the B/IP layer, opting for swapping different classes for different behavior, like BBMD and Foreign, so that contributes to your confusion!