JoelBender/bacpypes #505: asyncore removed from python 3.12

Mathadon, 2023-10-04:
Python 3.12 has been released on October 2nd. This version removed the `asyncore` module, which has been deprecated since python 3.6. This means bacpypes no longer works on python 3.12.

Would it be possible to update the implementation of bacpypes accordingly?

See python release notes:
https://docs.python.org/3/whatsnew/3.12.html?highlight=asyncore

bbartling, 2023-10-04:
Can you switch over to bacpypes3?

bbartling, 2023-10-04:
Its the new asyncio project: https://github.com/JoelBender/BACpypes3

Mathadon, 2023-10-04:
Oh I had no clue! 

I checked @JoelBender 's profile but it's not in the listed repositories:
[image]

I'll have a look at that, thanks!

closed 2023-10-04 by Mathadon