JoelBender/bacpypes #515: Clearing / Releasing a priority

kheldaroz-droid, 2024-01-11:
Hi Joel,

I am trying to release a Priority (eg priority 8 which is manual)
I can't find any examples

here is what I've been trying

`def release_bacnet_priority(device_instance, addr, object_type, priority=8):
    logging.info(f"Releasing Priority {priority} on Device {device_instance} at {addr} ({object_type})")

    # Adjust priority for zero-based indexing
    priorityToSet = priority - 1

    # Create an array of PriorityValue, all set to None
    priority_array_values = [PriorityValue(null=None) for _ in range(16)]

    # Set the priority you want to release to Null
    priority_array_values[priorityToSet] = PriorityValue(null=Null())

    # Debugging: Inspect the priority array values
    for idx, val in enumerate(priority_array_values):
        logging.info(f"Priority {idx + 1}: {val.dict_contents()}")

    # Convert the list into a PriorityArray object
    priority_array = PriorityArray(priority_array_values)

    # Debugging: Log the entire PriorityArray
    logging.info(f"Constructed PriorityArray: {priority_array}")

    # Create the write request
    request = WritePropertyRequest(
        objectIdentifier=(object_type, device_instance),
        propertyIdentifier='priorityArray',
        propertyValue=Any(priority_array),
        # No priority parameter since we're modifying the PriorityArray itself
    )
    request.pduDestination = Address(addr)
    iocb = IOCB(request)

    # Attach the address to the IOCB instance
    iocb.addr = addr
    iocb.object_type = object_type

    callback_function = lambda iocb, prop='priorityArray': on_property_release_response(iocb, prop)
    iocb.add_callback(callback_function)
    this_application.request_io(iocb)

def on_property_release_response(iocb, prop):
    if iocb.ioError:
        logging.error(f"{color.RED}Error Releasing to property {prop} on {iocb.addr} ({iocb.object_type}): {iocb.ioError}{color.END}")
    else:
        logging.info(f"{color.GREEN}Successfully Released Priority on property {prop} on {iocb.addr} ({iocb.object_type}){color.END}")`

this is my Error

`INFO:root:Releasing Priority 8 on Device 0 at 192.168.3.151 (analogOutput)
INFO:root:Priority 1: {}
INFO:root:Priority 2: {}
INFO:root:Priority 3: {}
INFO:root:Priority 4: {}
INFO:root:Priority 5: {}
INFO:root:Priority 6: {}
INFO:root:Priority 7: {}
INFO:root:Priority 8: {'null': <bacpypes.primitivedata.Null object at 0xffff83065150>}
INFO:root:Priority 9: {}
INFO:root:Priority 10: {}
INFO:root:Priority 11: {}
INFO:root:Priority 12: {}
INFO:root:Priority 13: {}
INFO:root:Priority 14: {}
INFO:root:Priority 15: {}
INFO:root:Priority 16: {}
INFO:root:Constructed PriorityArray: <bacpypes.basetypes.PriorityArray object at 0xffff83064ed0>
INFO:root:An error occurred in bacnet_write_release: missing choice of PriorityValue
`

any help would be appreciated

JoelBender, 2024-01-11:
It looks like you are writing the entire priority array in one shot when you should be writing a `Null` to the to the **Present_Value** property and providing the priority as a the parameter to the Write Property Request.  The **Priority_Array** is a read-only property.
```python
>>> request = WritePropertyRequest(destination=Address("1.2.3.4"),
...     objectIdentifier=('analogValue', 1), propertyIdentifier='presentValue',
...     propertyValue=Any(),
...     priority=8
... )
>>> request.propertyValue.cast_in(Null())
```
You really don't want to stomp on the priority array items that have been written by other, perhaps higher priority processes, and you don't want to have to read the whole thing then write it back out again for the same reason, so it's read-only.

kheldaroz-droid, 2024-01-11:
thanks Joel - that actually makes sense - maybe I'm getting the hang of python / BACnet

updated code for reference

`def release_bacnet_priority(device_instance, addr, object_type, priority=8):
    logging.info(f"Releasing Priority {priority} on Device {device_instance} at {addr} ({object_type})")

    # Create the write request
    request = WritePropertyRequest(
        objectIdentifier=(object_type, device_instance),
        propertyIdentifier='presentValue',
        propertyValue=Any(),
        priority=priority

    )
    request.propertyValue.cast_in(Null())
    request.pduDestination = Address(addr)
    iocb = IOCB(request)

    # Attach the address to the IOCB instance
    iocb.addr = addr
    iocb.object_type = object_type

    callback_function = lambda iocb, prop='presentValue': on_property_release_response(iocb, prop)
    iocb.add_callback(callback_function)
    this_application.request_io(iocb)

def on_property_release_response(iocb, prop):
    if iocb.ioError:
        logging.error(f"{color.RED}Error Releasing to property {prop} on {iocb.addr} ({iocb.object_type}): {iocb.ioError}{color.END}")
    else:
        logging.info(f"{color.GREEN}Successfully Released Priority on property {prop} on {iocb.addr} ({iocb.object_type}){color.END}")`

closed 2024-01-11 by kheldaroz-droid