JoelBender/bacpypes #521: COV subscription setting lifetime to 0 causes Error

kheldaroz402, 2024-03-14:
Hi All,
 I'm trying to talk to a Mitsubishi controller and it only supports COV subscription where lifetime is zero

when I try and do that, I get the following error

` File "/opt/WebAI/WebAIPy/BACnetIP.py", line 2087, in <module>
    @recurring_function(delay)
     ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/WebAI/WebAIPy/virtualenv/lib/python3.11/site-packages/bacpypes/task.py", line 253, in recurring_function_decorator
    task.install_task()
  File "/opt/WebAI/WebAIPy/virtualenv/lib/python3.11/site-packages/bacpypes/task.py", line 192, in install_task
    raise RuntimeError("interval must be greater than zero")
RuntimeError: interval must be greater than zero
`

I also tried setting it to none

class SubscriptionContext:

    def __init__(self, address, objid, confirmed=None, lifetime=None):
        if lifetime == 0:
            lifetime = None
        if _debug: SubscriptionContext._debug("__init__ %r %r confirmed=%r lifetime=%r", address, objid, confirmed, lifetime)
        global subscription_contexts, next_proc_id

        # destination for subscription requests
        self.address = address

        # assign a unique process identifer and keep track of it
        self.subscriberProcessIdentifier = next_proc_id
        next_proc_id += 1
        subscription_contexts[self.subscriberProcessIdentifier] = self

        self.monitoredObjectIdentifier = objid
        self.issueConfirmedNotifications = confirmed
        self.lifetime = lifetime

JoelBender, 2024-03-14:
Have you started with the **COVClientApp.py** sample application and you are expecting to call `send_subscription()` at some interval based on the lifetime stored in the context?

kheldaroz402, 2024-03-14:
Yes I think I did start with that app - it's been heavily modified since