JoelBender/bacpypes #526: Priority Array

kheldaroz402, 2024-05-31:
When I try and get the priority array to show up (say in yabe)  it just kills that type of object.

`def create_objects_database(app):
    mydbProgram = None
    mycursorProgram = None
    global objects
    logging.info(f"{color.PURPLE}************ Create the objects that hold the result values. ************{color.END}")

    object_creation_map = {
        'AnalogInput': create_analog_input,
        'AnalogOutput': create_analog_output,
        'AnalogValue': create_analog_value,
        'positiveIntegerValue': create_positive_analog_value,
        'BinaryValue': create_binary_value,
        'BinaryInput': create_binary_input,
        'BinaryOutput': create_binary_output,
        'Accumulator': create_accumulator,
        # 'Averaging': create_averaging,
        'PulseConverter': create_pulseConverter,
        'File': create_file,
        'MultiStateInput': create_multistate_input,
        'MultiStateOutput': create_multistate_output,
        'MultiStateValue': create_multistate_value,
        'CHARACTERSTRING_VALUE': create_characterstring_value,
        'Schedule': create_schedule,
        'Calendar': create_calendar,
        'NotificationClass': create_notification_class,
        'DATETIME_VALUE': create_datetime_value,
        'TIME_VALUE': create_time_value,
        'DATE_VALUE': create_date_value
    }

    try:
        mydbProgram = poolProgram.get_connection()
        mycursorProgram = mydbProgram.cursor(dictionary=True, buffered=True)

        mycursorProgram.execute("SELECT * FROM BACnet_Objects")
        myresult = mycursorProgram.fetchall()

        for row in myresult:
            logging.info(f"Object_Types_Name {row['Object_Types_Name']}")
            object_creation_function = object_creation_map.get(row['Object_Types_Name'])

            if object_creation_function:
                obj = object_creation_function(row)
                app.add_object(obj)
                objects[row['Object_Name']] = obj
                logging.info(f"Creating : {row['Object_Name']}. : {row['Object_Types_Name']}.")
            else:
                logging.info(f"Unknown Object Type: {row['Object_Types_Name']}. Skipping row.")

    except (MySQLError, Exception) as e:
        logging.info(f"An error occurred in create_objects_database: {e}")
    finally:
        if mycursorProgram:
            mycursorProgram.close()
        if mydbProgram:
            mydbProgram.close()`

`def create_analog_output(row):
    try:
        ArrayOfPropertyIdentifier = ArrayOf(PropertyIdentifier)
        property_list = ArrayOfPropertyIdentifier(['presentValue', 'statusFlags', 'eventState', 'outOfService', 'units', 'minPresValue', 'maxPresValue', 'resolution', 'covIncrement', 'description', 'highLimit', 'lowLimit', 'limitEnable','priorityArray'])
        AnalogOutputObject(
            objectName=row['Object_Name'],
            objectIdentifier=(AnalogOutputObject.objectType, row['Object_Identifier']),
            presentValue=row['Present_Value'],
            statusFlags=calculate_status_flags(row),
            eventState=row.get('Event_State', 0),
            outOfService=True if row['Out_Of_Service'] == 1 else False,
            units=row['BACnet_Engineering_Units'],
            minPresValue=row['Min_Pres_Value'],
            maxPresValue=row['Max_Pres_Value'],
            resolution=row.get('Resolution', None),
            covIncrement=row.get('COV_Increment', 1),
            description=row['Description'],
            highLimit=row.get('High_Limit', None),
            lowLimit=row.get('Low_Limit', None),
            limitEnable=row.get('Limit_Enable', [False, False]),
            propertyList=property_list,
        )
# Initialize the priority array with None values
        priority_array = PriorityArray([None] * 16)

        # Set the Present_Value at the specified priority position if within range
        if 'priority' in row and 1 <= row['priority'] <= 16:
            priority_index = row['priority'] - 1  # Adjust for 0-based index
            priority_array[priority_index] = row['Present_Value']

        # Assign the priority array to the object
        AnalogOutputObject.priorityArray = priority_array

        logging.debug(f"Created AnalogInputObject: {AnalogOutputObject}")

        return AnalogOutputObject

    except Exception as e:
        logging.error(f"An error occurred in create_analog_input: {e}")
        return None`

this stops all objects of analog_output, if I remove references to priority array it works fine

JoelBender, 2024-05-31:
You are expecting `[None]` to be "cast" as `[PriorityValue(None)]` but that isn't provided, `PriorityValue` is a `Choice` with no element selected and there needs to be at least one.  If you specified `[PriorityValue()] * 16` you would eventually be tripped up because it would be the same item repeated 16 times, and the same for `[PriorityValue(null=())] * 16`.

The definition of `PriorityArray` includes a "prototype" value that sets the `null` option and this is copied for each item of the array when an instance is created so to create a new empty array, just call `PriorityArray()`.

closed 2024-05-31 by JoelBender

kheldaroz402, 2024-06-03:
thanks - that got that sorted

`def create_analog_output(row):
    try:
        ArrayOfPropertyIdentifier = ArrayOf(PropertyIdentifier)
        property_list = ArrayOfPropertyIdentifier([
            'presentValue', 'statusFlags', 'eventState', 'outOfService', 'units',
            'minPresValue', 'maxPresValue', 'resolution', 'covIncrement', 'description',
            'highLimit', 'lowLimit', 'limitEnable'
        ])

        analog_output = AnalogOutputObject(
            objectName=row['Object_Name'],
            objectIdentifier=(AnalogOutputObject.objectType, row['Object_Identifier']),
            presentValue=row['Present_Value'],
            statusFlags=calculate_status_flags(row),
            eventState=row.get('Event_State', 0),  # Assuming default value is 0
            outOfService=bool(row['Out_Of_Service']),  # Convert to bool
            units=row['BACnet_Engineering_Units'],
            minPresValue=row['Min_Pres_Value'],
            maxPresValue=row['Max_Pres_Value'],
            resolution=row.get('Resolution', None),
            covIncrement=row.get('COV_Increment', 1),
            description=row['Description'],
            highLimit=row.get('High_Limit', None),
            lowLimit=row.get('Low_Limit', None),
            limitEnable=row.get('Limit_Enable', [False, False]),
            propertyList=property_list,
        )

        # Initialize the priority array with default values
        priority_array = PriorityArray()

        # Set the Present_Value at the specified priority position if within range
        if 'Priority' in row and 1 <= row['Priority'] <= 16:
            priority_index = row['Priority']   # Don't need to Adjust for 0-based index
            priority_array[priority_index] = PriorityValue(real=Real(row['Present_Value']))

        # Assign the priority array to the object
        analog_output.priorityArray = priority_array

        # Print out the priority array
        logging.info(f"Priority Array Contents: Current Priority is {row['Priority']} with a value of {row['Present_Value']}")
        # for i, priority_value in enumerate(priority_array):
        #     # Log the actual value contained in the PriorityValue object
        #     if priority_value:
        #         logging.info(f"Priority {i+1}: {priority_value.real}")
        #     else:
        #         logging.info(f"Priority {i+1}: None")

        logging.debug(f"Created AnalogOutputObject: {analog_output}")

        return analog_output

    except Exception as e:
        logging.error(f"{color.RED}An error occurred in create_analog_output: {e}{color.END}")
        return None`