JoelBender/bacpypes #532: [HELP] Unable to detect multiple devices connected to the same panel address

knityanand, 2024-08-03:
[image]

Yabe shows two devices
Device 4011 - 0.0.0.0:4011 via 192.168.1.12:47808
Device 4012 - 0.0.0.0:4012 via 192.168.1.12:47808

Am unable to access these two devices and read parameters from them. WhoAmI is returning these two devices. 

How can I access the same? Any help is greatly appreciated.

knityanand, 2024-08-05:
Thanks for your response.

Tried the same as suggested by you. I get the error as shown in the attachment below.
[bacpypes3.txt](https://github.com/user-attachments/files/16491895/bacpypes3.txt)

However, I did check wireshark to see what data is being returned when YABE sends whoami. I see the below info which seems to indicate that your info that 192.168.3.12 seems to be a BACnet router.

Building Automation and Control Network NPDU
    Version: 0x01 (ASHRAE 135-1995)
    Control: 0x28
        0... .... = NSDU contains: BACnet APDU
        .0.. .... = Reserved: Shall be zero and is zero.
        ..1. .... = Destination Specifier: DNET
        ...0 .... = Reserved: Shall be zero and is zero.
        .... 1... = Source specifier: SNET
        .... .0.. = Expecting Reply: Other than a BACnet-Confirmed-Request-PDU
        .... ..0. = Priority: Not a Life Safety or Critical Equipment message.
        .... ...0 = Priority: Normal message
    **Destination Network Address: 65535
    Destination MAC Layer Address Length: 0 indicates Broadcast on Destination Network
    Source Network Address: 70
    Source MAC Layer Address Length: 6
    SADR: 00:00:00_00:0f:ab (00:00:00:00:0f:ab)
    Hop Count: 254**

JoelBender, 2024-08-06:
The error comes from not knowing your local broadcast address.  You can install [ifaddr](https://pypi.org/project/ifaddr/) and let it figure it out from your interfaces, or you can give it your local IPv4 address and subnet in CIDR notation:
```shell
$ python3 -m bacypes3 --address 192.168.3.99/24
```
So you are correct, sending a local broadcast Who-Is-Router-To-Network should get a response:
```shell
> wirtn *
```
And a _global broadcast_ Who-Is will show the other devices:
```shell
> whois *:* 4000 4999
```
The address that comes back is the destination address for your other requests:
```shell
> read 70:0x000000000fab device,4011 object-name
```
Note that the MAC address is 6 octets long, so all of those zeros are significant.

knityanand, 2024-08-06:
Thx. This was very helpful... I got the below output. Am able to read param values now...

D:\SEIOT\bacpypes> python -m bacpypes3 --address 192.168.3.2/24
> wirtn *
192.168.3.12
    70
192.168.3.227
    50
> whois *:* 4000 4999
4010 70:0x000000000faa
4011 70:0x000000000fab
4001 192.168.3.23
> read 70:0x000000000fab binaryInput,11 presentValue
active

I tried checking this in bacpypes as well and it did parse the address. We are using bacpypes in our project. Could you point me to any documentation for the INI file or share a sample file where I can configure all these parameters? I would like to use the samples/ReadProperty.py for the same? Am I headed in the right direction?

Thanks in advance

knityanand, 2024-08-07:
Am able to read data from all devices now. Thx so much @JoelBender

closed 2024-08-07 by knityanand